Opentrons/opentrons #2948: Labware calibration: App changes for change in calibration point for troughs

sanni-t, 2019-01-22:
## overview

protocol_api2 now uses the well's center point as the calibration point for troughs. Specifically, now pipette's center should align with trough well center.  So we need to update the app's calibration step to show this new calibration point

## current behavior

Currently the calibration point for troughs is a point along the upper edge of the first well of a trough

## expected behavior

The image on labware calibration step should show a point in the center of the first well for calibration. In case of multichannels, the pipette should center over the trough well

btmorr, 2019-08-01:
cc @pantslakz

howisthisnamenottakenyet, 2019-08-01:
@btmorr @sanni-t Is this right?

[image]

btmorr, 2019-08-02:
Left graphic looks right. Right graphic is for calibrate-to-bottom. I believe this card is supposed to be for calibrate to top.

howisthisnamenottakenyet, 2019-08-02:
@btmorr Ok! If you confirm the design in the zeplin link then I'll remove the design tag from the ticket.

Kadee80, 2019-08-02:
@howisthisnamenottakenyet RE: calibrate to bottom image. We will need that as well since we swap out images/text based on if that advanced setting is turned on

howisthisnamenottakenyet, 2019-08-02:
updated

PR #3823 (2019-08-06): refactor(app): Display pipette in center of trough wells on apiv2
APIv2 uses the center of trough wells for calibrating single-channel pipettes.
This commit

- Adds an api_level member of the robot session. This is set here because in the
  future it will be read from the protocol itself, so we shouldn't just use the
  app's knowledge of the robot feature flags
- Modifies the confirm position components to read that member and use a
  different image if we're using a reservoir and apiv2.

Closes #2948

## Testing

Check all the following using any protocol that uses a `reservoir` labware and a [... cut]
commit f1c9d3af (2019-08-06): refactor(app): Display pipette in center of trough wells on apiv2 (#3823)

APIv2 uses the center of trough wells for calibrating single-channel pipettes.
This commit

- Adds an api_level member of the robot session. This is set here because in the
  future it will be read from the protocol itself, so we shouldn't just use the
  app's knowledge of the robot feature flags
- Modifies the confirm position components to read that member and use a
  different image if we're using a reservoir and apiv2.

Closes #2948
closed 2019-08-06 by sfoster1