Opentrons/opentrons #2987: Position of labware tube-rack-15_50ml  is wrong

simonchih, 2019-01-30:
## overview
Position of well 'C3' in labware tube-rack-15_50ml is NOT correct.

## current behavior
With version 3.6.5. Position of well 'C3' in labware tube-rack-15_50ml is NOT in the center of tube.

## steps to reproduce
1. Open OT2 App
2. Upload OmegaMagBind.test.ot2_after_26_part2.py below:

https://drive.google.com/open?id=1XI8PGR9ZVvDnVa2EGlzYQ0doRwo_8m_M

3. Calibrate every labware include tube-rack-15_50ml

4. Run OmegaMagBind.test.ot2_after_26_part2.py
==> Position of well 'C3' in labware tube-rack-15_50ml is NOT in the center of tube.
See video below:

https://drive.google.com/open?id=1xXcW6IwW1sBFAdSdY1O3we6bN1K9vVF4

## expected behavior
 Position of well 'C3' in labware tube-rack-15_50ml is in the center of tube.

btmorr, 2019-01-30:
@simonchih thanks for the bug report, and the protocol and video. 

Can you also provide the version numbers of your App and API? It's possible that this is a bug we've already fixed--I'll investigate, but it might be fixed by: 
- updating the software
- doing a factory reset of your labware calibrations

To reset the calibrations, open the App, click on your robot, and scroll down to the "Factory Reset" section--this will clear all of your calibrations, but will also fix this issue if it was caused by a bug in a prior version of the software.

mcous, 2019-07-17:
We've rewritten this definition in the `3.10.0` release and any positional problems at the definition level should have been solved. Please use the following definitions:

- Acrylic 15/50 rack: `opentrons_10_tuberack_falcon_4x50ml_6x15ml_conical_acrylic`
- [4-in-1 Tuberack Set](https://shop.opentrons.com/collections/racks-and-adapters/products/tube-rack-set-1): `opentrons_10_tuberack_falcon_4x50ml_6x15ml_conical`
    - <https://labware.opentrons.com/opentrons_10_tuberack_falcon_4x50ml_6x15ml_conical>

closed 2019-07-17 by mcous