Opentrons/opentrons #3045: Pipette Config: Number of Presses for Tip Pickup (Backend)

IanLondon, 2019-02-13:
## overview

As a user, I would like to set the number of presses for tip pickup for each of my pipettes.

## Acceptance Criteria

Add `numPressesTipPickup` to the fields in `GET /settings/pipettes` and `PATCH /settings/pipettes/$id/fields` endpoints. It is a unitless integer field `{"units": null, "type": "integer"}` with a minimum of 1.

`pick_up_tip` function will use, in order of priority:
1. `presses` function argument (eg, `pick_up_tip(presses=4)`)
2. pipette config for `numPressesTipPickup` under the pipette's ID, if `presses` arg was not specified
3. default value (3) if there is no `numPressesTipPickup` saved for that pipette

btmorr, 2019-02-14:
@umbhau @pantslakz please provide context on user impact

pantslakz, 2019-02-20:
Double confirmed with Ojas and David - number of presses seems to not make any difference in pipette performance, so we can remove this. @umbhau

closed 2019-02-20 by IanLondon