Opentrons/opentrons #3062: Thermocycler: Run App: Module Card Open/Close

umbhau, 2019-02-18:
As a Run App user, I would like to be able to open or close my Thermocycler from it's module card on the 'pipettes and modules' page, so that I can control it without an active protocol running. 

## Acceptance Criteria
- [ ] Button(s) or other affordance on the Thermocycler module card on the 'pipettes and modules' page to open and close thermocycler
- [ ] Button inactive if protocol is actively running
- [ ] Button inactive if TC is above safe temperature (TBD)
- [ ] Tooltips informing user why button is inactive 

## Open Questions / Notes
- TBD if/how we want to handle safe temps
- Story assumes there is not a user-facing HW button to open or close the TC; think it is still valid if we decide to add one, but worth a discussion
- Story is not strictly necessary for science testing. In MVP but lower priority. 

## Design
- TBD @pantslakz 
- @pantslakz I wrote it up as inactive buttons + a tooltip, but open to other options for handling those two scenarios.

pantslakz, 2019-08-01:
IMO this feature is not necessary as user can rely on physical button on module.

But if need be: https://zpl.io/boLe8WG

umbhau, 2019-08-06:
Closing; will not add a software button to open/close lid for now (except as part of specific user flows)

closed 2019-08-06 by umbhau