Opentrons/opentrons #3081: [edge] liquid tracking broke a bit, may be related to mix

howisthisnamenottakenyet, 2019-02-19:
##Rows 3-5 in QA doc

Example protocol, Look at Transfer 4: https://drive.google.com/open?id=1bJ_6abXSqM8f03nDCWSK-uNfJa72UwiD

[image]

Aspirating 1 uL from a well with 297 uL in it generates the "not enough liquid in well" warning. 

Resulting well goes from empty to half green liquid half unknown, 2uL total volume. If you duplicate the step then only the first attempted aspirate from B1 generates that error. If you switch the duplicate and original which ever happens first has the error.

IanLondon, 2019-02-20:
We talked about this in person: the problem is that the "inner mix" in that Transfer is actually over-aspirating, since there isn't 2uL of liquid in the well. 

We need to make a software change to allow the timeline warnings to differentiate between the "inner mix"'s aspirate vs the main transfer aspirates to be able to deliver different warning messages for those 2 contexts

howisthisnamenottakenyet, 2019-02-20:
@IanLondon I'll put a design tag on this ticket and add it to the ux queue

closed 2022-09-30 by bndo