Opentrons/opentrons #3107: Thermocycler: Implement Spanning

Laura-Danielle, 2019-02-21:
## overview

Related to issue #2962 . The Thermocycler is our first module to span multiple slots in a robot. We need to have an elegant way to handle this concept. Once this is implemented, it should be able to be 
applied to any labware that may span multiple slots.

## current behavior

- Currently there is no way for the robot to account for spanning. It simply relies upon the user to choose the correct slot placement.

## expected behavior

- Robot should remove any slots that would be covered by spanning based on labware dimensions and the input slot

IanLondon, 2019-02-21:
@Laura-Danielle Please keep me & Brian in the loop so we don't make a roadblock for how PD handles slots!

btmorr, 2019-02-28:
@umbhau please move to post-MVP milestone

kinnari-opentrons, 2019-09-03:
Attaching this protocol for reference. The p20 pipette has to clear screw cap tubes in the 2ml aluminum block on the the temperature module and the concern about clearance was raised. @TaylorGoodman

kinnari-opentrons, 2019-09-03:
[Swift_2S_Turbo_V2_TC_test.txt](https://github.com/Opentrons/opentrons/files/3570544/Swift_2S_Turbo_V2_TC_test.txt)

commit 37a4e115 (2019-10-01): feat(api): implement deck item spanning first pass

During simulation, prevent deck items from being loaded into slots obscured by other deck items,
including the thermocycler, which spans multiple slots.

Closes #3107
PR #4160 (2019-10-02): feat(api): implement deck item spanning first pass
## Overview

During simulation, prevent deck items from being loaded into slots obscured by other deck items,
including the thermocycler, which spans multiple slots.

Closes #3107

## Review Requests

 - [ ] upload a protocol with thermocycler and other non-overlapping deck items, should simulate
- [ ] upload a protocol with thermocycler and deck items in 7, 8, 10 or 11, should raise exception
- [ ] upload a protocol without modules with two items in same slot, should still raise exception
- [ ] upload a protocol without modules and other non- [... cut]
commit 1b621a42 (2019-10-02): feat(api): implement deck item spanning first pass (#4160)

During simulation, prevent deck items from being loaded into slots obscured by other deck items, including the thermocycler, which spans multiple slots.

Closes #3107
closed 2019-10-02 by b-cooper