Opentrons/opentrons #3175: Robot should not use random MAC  addresses

sfoster1, 2019-03-06:
As a user who needs to enter the MAC address of my robot into my institution's MAC whitelist to connect to the internet, I need that address to not randomly change.

Success criteria:
- The MAC address presented to the app and to the base station is the hardware address of the interface
- The MAC address presented to the app and to the base station does not change with time or with robot reboots

kentcr, 2019-08-20:
I was told by support that this issue "should be fixed soon I believe." I can't tell if that means someone is already working on it. Is someone? If not, I might have time to look into it.

kentcr, 2019-08-20:
Adding this to the NetworkManager config fixes it:
```
[device]
wifi.scan-rand-mac-address=no
```
e.g. /etc/NetworkManager/conf.d/00-fixmac.conf

mcous, 2019-08-28:
Fixed by the buildroot switch in 3.11.x

@kentcr the latest version of the software should resolve this issue

closed 2019-08-28 by mcous