Opentrons/opentrons #3194: Add serialization to liquid tracking

howisthisnamenottakenyet, 2019-03-11:
As a user I would like to be able to track the samples #s I placed in my starting deck state

## Background
Serialization number currently is only visible when you're placing liquid into labware in the starting deck state. It does not display in any liquid tracking tooltips. 

## Acceptance criteria
- [ ] In liquid tracking tooltips liquids with a serialization # should display that #.

IanLondon, 2019-03-18:
AFAIK we still haven't answered these questions:

- How are serialization numbers assigned within a labware, and across labware?
- How can users change the serialization order within a labware and across labware?
- What happens when you check "serialize" on a liquid that is already on the deck?
- If you duplicate a labware that contains serialized liquids, do the serialize numbers increment or duplicate?
- Can you have multiple instances of a liquid with the same serialization number across labware? What about within a labware?
- If you un-check "serialize", save that, then re-check it and save that, did it "reset" the serialization order?

Those questions seem pretty involved to answer, and to implement! A potential simple & expedient approach for the next iteration could be something like: 

PD controls serialization order. Its rule is:
* Liquids are serialized within a labware from top to bottom then left to right.
* All serialized liquids, in each labware, start at 1. This means that you cannot (for this iteration) serialize across labware. If you have 4 plates with A1 A2 A3 filled with serialized "Sample", it will always be `A1: Sample 1, A2: Sample 2, A3: Sample 3` in each labware.
* If you have multiple ingredients in a labware, each of them starts at 1. So column 1 might look like: Sample 1, Sample 2, Buffer 1, Buffer 2, Sample 3, Sample 4.

If we have that limitation in place, we get these answers to the questions:
* Users can't change the serialization order of initial deck setup
* Liquids can have the same serialization number across labware, but not within a labware (eg there can be a "Sample 1" in Trough and in 96-Flat, but only 1 "Sample 1" in the Trough, you can't fill multiple wells with "Sample 1")
* If you duplicate a labware, it'd be the same end result as if you make a new labware and fill the same wells with the same liquid.
* If you check "Serialize" for an un-serialized liquid already on the deck, PD assigns the serialization numbers following the "top to bottom then left to right, start at 1"
* Similarly, un-checking and re-checking the "Serialize" checkbox will always end up with your liquid setup in the same state it was.
* These assignments I'm talking about only happen in **initial well setup**. Once you move the liquids around, their serialization number follows them - so if you transfer source plate to empty dest plate, Sample 1 from A1 to dest C1 and Sample 2 from A2 to dest B1, dest plate will have `B1: Sample 2, C1: Sample 1`

IanLondon, 2019-03-18:
I missed: what happens when you combine liquids in these cases:
* Sample 1 + Sample 1
* Sample 1 + Sample 2
* Sample 1 + DNA 2

closed 2022-09-30 by bndo