Opentrons/opentrons #3241: Labware Library: labware definitions should have well bottom shape

mcous, 2019-03-21:
## overview

As a labware library user, I would like to know the shape of the bottom of a labware's wells so I can identify the correct definition for my labware.

## acceptance criteria

- [ ] Existing labware definitions contain enum identifying well bottom shape
- [ ] Generator updated

## implementation details / questions

- Should this be added per labware or per well?
    - Do labware exist with differing well bottoms?
- Is this metadata? We're not trying to do anything functional here
- Possible values:
    - Flat `|_|`
    - Round `U`
    - Pointed `V`
    - Other?

Laura-Danielle, 2019-03-22:
1. Yes labware exist with differing well bottoms (and they would be considered irregular)
2. If your question asking if it's metadata means you are wondering whether it should be required or not, I do not think it should _not_ be required for a labware definition personally unless we are planning on adding functionality to the robot from this information.
2a. If we are not going to be using any of this info I think we should add the well bottom information as a tag and/or key on metadata to leave the door open for us to actually add well bottom geometry to a well.

IanLondon, 2019-03-22:
I think it should def be marked as metadata to discourage trying to do anything functional with it.

> Do labware exist with differing well bottoms?

As long at tube racks are a definition, this is a common case. Otherwise - I don't know how common it would be for "static" labware where you can't swap tubes / PCR strips / etc, but it's _possible_, especially for crazy crystallography plates etc.

Since bottom shape is just metadata and not geometry, whether or not it actually matters to represent differing well bottoms depends on the design requirements for how we want to display/sort/search the weird ones. Tube racks with mixed tubes are something to think about, IDK how we're doing those in LL.

AFAIK "pointed" is usually called "v-bottom" "Round bottom' and "flat bottom" are things I've seen.

howisthisnamenottakenyet, 2019-03-22:
May want to differentiate between V-bottom and Conical bottom as came up on slack. 

AFAIK the main functionality add we get out of knowing the well bottom shape is having a better/more specific guide on how to measure labware.

IanLondon, 2019-03-22:
Do labware manufacturers use "conical bottom"? Eg https://ecatalog.corning.com/life-sciences/b2c/US/en/Microplates/Assay-Microplates/96-Well-Microplates/Falcon%C2%AE-96-well-Polypropylene-Storage-Plates/p/353263 it's "conical" but called "v-bottom"

Laura-Danielle, 2019-03-22:
This is V-shaped: https://www.usascientific.com/25-ml-graduated-reservoir-white-bulk.aspx

This doesn't even describe the well shape: http://www.bio-rad.com/en-us/sku/hss9601-hard-shell-96-well-pcr-plates-high-profile-semi-skirted-clear-clear?ID=HSS9601

It only has the geometry of it.

IanLondon, 2019-03-22:
Tubes, esp centrifuge tubes, are commonly called "conical tubes" https://www.fishersci.com/shop/products/falcon-15ml-conical-centrifuge-tubes-5/p-193301

But can also be called "v-bottom tubes" https://www.fishersci.com/shop/products/2-0-ml-v-clear-tube-500-pk/501251295#?keyword=v-bottom+tube

Laura-Danielle, 2019-03-22:
> Do labware manufacturers use "conical bottom"? Eg https://ecatalog.corning.com/life-sciences/b2c/US/en/Microplates/Assay-Microplates/96-Well-Microplates/Falcon%C2%AE-96-well-Polypropylene-Storage-Plates/p/353263 it's "conical" but called "96-well Clear V-Bottom"
> 
> And this trough isn't "conical" but is also called "v bottom" https://cisnovo.com/product/disposable-solution-reservoirs-ps-50-mldivided-trough/
> 
> > Tapered “V” bottoms

@IanLondon I'm not sure we can categorize that a v-bottom based on your example. When I checked out the technical specs listed for that labware it only had flat bottom shapes called out. Not every manufacturer lists out well shape as an adjective as I pointed out above. Sometimes it is simply a diagram/technical drawing.

IanLondon, 2019-03-22:
I just want to get across that in the outside world, "v-bottom" is generic; it is applied to troughs, circular wells, and tubes.

If we're doing an illustrative icon then hopefully that will help users pick up on the difference between "v-bottom (trough-like)" vs "conical"

howisthisnamenottakenyet, 2019-03-22:
Oh cool. So V, flat, round, and....other?

mcous, 2019-04-16:
Talked to @IanLondon today about this, and we have a proposal for resolving this ticket:

- Add a top level `grids` field to labware definitions that is an array of objects
- An individual object in the `grids` array has two fields to start
    - `wells: Array<string>` - All the well IDs in the grid
    - `metadata: Object` - Metadata about each well in the grid
- For this ticket, the `metadata` object would be `{wellBottomShape: 'flat' | 'round' | 'v'}`

Out of scope for this ticket, but implied future work:

- Explore moving other well properties, like `totalLiquidVolume` and well shape to `grids`
- Future well-specific render stuff can go in `grids[].metadata`

Example (For `opentrons_6x15_ml_4x50_ml_tuberack`):

```js
{
  "ordering": [
    ["A1", "B1", "C1"],
    ["A2", "B2", "C2"],
    ["A3", "B3"],
    ["A4", "B4"]
  ],
  "grids": [
    {
      "wells": ["A1", "B1", "C1", "A2", "B2", "C2"],
      "metadata": {
        "wellBottomShape": "v"
      }
    },
    {
      "wells": ["A3", "B3", "A4", "B4"],
      "metadata": {
        "wellBottomShape": "v"
      }
    }
  ],
  // ...
}
```

Laura-Danielle, 2019-04-16:
I like this. Seems simple, and we can build things into it as we learn more info on data needed to fully describe a labware def :)

PR #3490 (2019-05-29): refactor(shared-data): Add well groups with metadata to definitions2
## overview

~Blocked by #3489. Will rebase this PR over `edge` once 3489 is merged~ Rebased and ready for review

This PR adds per-well-group metadata to all labware definitions and closes #3241. The shape of a well group is:

```js
export type LabwareBrand = {|
  brand: string,
  brandId?: Array<string>,
  links?: Array<string>,
|}

export type LabwareWellGroupMetadata = {|
  displayName?: string,
  displayCategory?: LabwareDisplayCategory,
  wellBottomShape?: WellBottomShape,
|}

export type LabwareWellGroup = {|
  wells:  [... cut]
commit aa37f828 (2019-05-29): refactor(shared-data): Add well groups metadata to definitions2 (#3490)

closes #3241
closed 2019-05-29 by mcous