Opentrons/opentrons #3248: Calibration w/ Multichannel & Single Channel

pyeguy, 2019-03-22:
## overview
It seems that when using a single channel and multi channel pipette in the same protocol the labware calibration is always performed using the single channel pipette. Unfortunately this often causes the multichannel to fall out of calibration particularly when using 384 well plates

## current behavior
The current behavior seems to use the single channel to calibrate labware preferentially regardless of which side the pipette is mounted or the order of the tip racks. Unfortunately it would appear that the calibration of the single channel is insufficient to correctly calibrate the entire arm. This is evidenced by large (~0.5mm) deviations by the multichannel pipette mounted in a different position from the center of well as calibrated by the single channel. 

## steps to reproduce

Cue up a protocol w/ a single and multichannel mounted arm and proceed to calibration. Calibration will take place w/ single channel. Then use the multichannel to (attempt) pipette in to 384 well plate and tips will catch on the sides of wells often causing bent or misaligned tips to fail. 

## expected behavior

I think calibrations should be specific to the pipette being used if the tolerances of the device are low enough that using a single arm (ie left or right) to designate the labware calibration results in improper alignment for the other position pipette.

btmorr, 2019-05-06:
Hi @pyeguy, thanks for writing this up. We're planning on re-working how labware calibration is performed. I'll make sure that this write-up is included in the new design. (cc @pantslakz @umbhau)

mcous, 2019-08-29:
I'm going to close this ticket because this sounds like a problem with your robot's pipette offsets. If you're still experiencing these issues, our support team (<https://support.opentrons.com> and/or in-app chat) can help you correct this issue.

Your feedback is very much appreciated, though! We've incorporated it into our plans for our longer term fixes to how our software handles these offsets and its calibration processes. I'll post a link the milestone in this thread for tracking purposes once the work has been spec'd out

closed 2019-08-29 by mcous

pyeguy, 2019-08-29:
> Closed #3248 (https://link.getmailspring.com/link/4AC0173B-4332-4709-BB96-AD5C953B8E28@getmailspring.com/0?redirect=https%3A%2F%2Fgithub.com%2FOpentrons%2Fopentrons%2Fissues%2F3248&recipient=cmVwbHkrQUJJTjJFQklEU0dXTFY1M01USDRSUzUzT1VaMjVFVkJOSEhCU1NYSzJFQHJlcGx5LmdpdGh1Yi5jb20%3D).

Great, thanks for the feedback!

I will note that it looks like things are behaving better as of latest updates.
Best,
Cameron