Opentrons/opentrons #3344: Pipette 1.5: Add P10M v1.5

sfoster1, 2019-04-11:
The new pipette model for the P10 multi needs to be added to pipetteModelSpecs.json. It should have the same behavior as the P10M 1.4, but with a different aspiration function which will be provided by the hardware team.

The serial number format for the p10m 1.5 is P10MV15ddmmssss

Done is:
- Pipettes flashed with a matching serial are detectable in the api and app as 1.5
- write_pipette_memory.py correctly reads barcodes and flashes pipettes with the 1.5 barcode

TaylorGoodman, 2019-04-19:
Below is the updated P10M function as of 4/19/19.
min uL | max uL | slope | intercept
0.9 | 1.774444444 | -0.1917910448 | 1.2026
1.774444444 | 2.151481481 | -0.0706286837 | 1.0125
2.151481481 | 2.898518519 | -0.04343083788 | 0.9540
2.898518519 | 6.373333333 | -0.00905990194 | 0.8544
6.373333333 | 11.00259259 | -0.002325900358 | 0.8115

btmorr, 2019-04-22:
The minimum step resolution for the plunger motor is 0.04mm, so this PR should also change the discontinuity test from 1e-2 to 2e-2.

commit d2657002 (2019-04-22): feat(api): add P10M 1.5 config data

closes #3344
PR #3365 (2019-04-22): feat(api): add P10M 1.5 config data
## overview
- add P10M v1.5 config data
- increase `test_ul_per_mm_continuous` difference limit by 1 mm

closes #3344
closed 2019-04-22 by ahiuchingau