Opentrons/opentrons #3381: API: Recover from bad deck calibration data

Laura-Danielle, 2019-04-23:
## overview
When a robot has bad deck calibration data which causes a limit switch error, a user cannot complete deck calibration through the app.

## steps to reproduce
- Add this deck calibration data to your robot and attempt to calibrate:
```
{
    "gantry_calibration": [
        [
            0.99778430477,
            0.00127912359,
            0.0,
            -0.1996993104
        ],
        [
            -0.01861990175,
            0.99771987207,
            0.0,
            2.19834795825
        ],
        [
            0.0,
            0.0,
            1.0,
            -14.6
        ],
        [
            0.0,
            0.0,
            0.0,
            1.0
        ]
    ]
}

```

## expected behavior

Robot should fall-back to known "good enough" deck calibration data to allow the user to complete the flow.

mcous, 2019-06-17:
Closed by #3550

closed 2019-06-17 by mcous

wab8, 2019-08-14:
Hi Laura, 

I am a user and currently my OT-2 is having bad calibration problem. Do you mind telling me where and how I can put the gantry data for my OT-2? 

Thanks,
Wihan

Marie201293, 2022-09-01:
Hello, 

I have the following problem. I have taken over an Opentrons internally in the company and reset the calibrations etc. to factory settings. Now I want to do a deck calibration, but from deck 1 the tip hits the edges of the seats and goes too far down. What can I do to calibrate the deck?
Thanks a lot