Opentrons/opentrons #3384: Execute python script via SSH, robot won't proceed moving

ffffffrank, 2019-04-24:
## overview

## steps to reproduce

The "from opentrons import robot, labware, instruments" should work remotely via ssh in Paramiko.

sfoster1, 2019-04-29:
@ffffffrank Our software relies on a number of environment variables that are only loaded when you start a login shell on the robot. Try changing the argument to `ssh_client.exec_command` to be `'bash -lc "python3 /data/test.py"'`

ffffffrank, 2019-04-29:
> @ffffffrank Our software relies on a number of environment variables that are only loaded when you start a login shell on the robot. Try changing the argument to `ssh_client.exec_command` to be `'bash -lc "python3 /data/test.py"'`

OMG, that works perfectly! Thank you very much!

closed 2019-04-30 by ffffffrank