Opentrons/opentrons #3432: Pipette+: A loaded protocol with old constructors, but new pipettes attached, should throw a warning/show a window that gives suggestion to update

Laura-Danielle, 2019-05-09:
## overview
1. App should be able to detect that an uploaded protocol has the incorrect pipette attached when new constructors are used for _old_ pipettes and not allow the protocol to proceed.
2. App should display a window if new pipettes are attached but old constructors are used. The window should say that pipette+ has a larger range of volumes but older volumes are being used due to old constructors in protocol. This should link to a support doc on how to update constructors and who to contact if there are issues.

umbhau, 2019-06-18:
Decided in estimation to display an alert modal for all protocol <> robot pipette incompatibilities, not just gen2/gen1 issues.

FYI @pantslakz

mcous, 2019-10-02:
@umbhau @pantslakz I have a few questions about this ticket:

- Why are we adding yet another modal instead of augmenting our existing warning section for pipette incompatibilities?
- Why are we allowing the user to continue if we know the pipettes are not compatible?

btmorr, 2019-10-24:
Closed by #4185

closed 2019-10-24 by btmorr