Opentrons/opentrons #3465: RFC: Bundling data in python protocols

sfoster1, 2019-05-20:
## overview

Customers want to bundle additional data with their protocols, for instance csv files determining amounts to aspirate from other analysis tools or custom Python modules so they can factor their code into reusable units. In addition, #3454 would be simplified if it was easier to bundle v2 labware definitions along with python protocols. This is a proposal about how to accomplish this bundling: have a zip file with exactly one python file at the top level and arbitrary other contents.

## bundle file format

A *protocol bundle* would be a zip file containing
- precisely one file at the top level with a filename ending in `.py`, which will be executed by the API server. The name of the file is the name surfaced by the API server as the name of the protocol. This is the entry point and probably main body of the protocol.
  - note that other python files may be present elsewhere in the zip file's hierarchy
- other files basically without restriction
- file paths in the zip file will be sandboxed to the temporary directory: absolute paths will be interpreted as if the root were the protocol's temporary directory

## bundle file handling

When the API server is sent a protocol bundle, it will unzip it into a temporary directory and execute the file in the temporary directory with 
- `os.cwd()` set to the temporary directory
- the temporary directory added to `sys.path`
- [maybe] the temporary directory added to `LD_PRELOAD`

Labware bundled with the protocol will live in an implicit `protocol` namespace. Calls to `labware.load` with an unspecified namespace will prefer this namespace. see #3454 

That means the python code executed for the protocol can
- open files in the current directory (e.g. if a file `aspirate.csv` is bundled, the protocol can do `open('./aspirate.csv')`
- import python modules present in subdirectories
- run executables or use shared object libraries in the subdirectories

## runapp ux

To create the zipfiles, the runapp should be able to use a filesystem dialog to select multiple files or directories and zip them into a zipfile. The robot API server can handle the validation if necessary. In addition, the runapp could support uploading a zip file directly.

IanLondon, 2019-05-20:
> Labware bundled with the protocol will live in an implicit protocol namespace. Calls to labware.load with an unspecified namespace will prefer this namespace.

1. Could `labware.load` default to `opentrons` namespace? I'm assuming the standard opentrons defs will continue to be stored on the robot and won't be required in the bundle.

2. I don't understand having `"protocol"` as a literal namespace. I bundle in the def `cool_labware.json` which has a `"namespace": "my_lab"` in it -- where does `"protocol"` namespace fall in there? To get the intended behavior that I think you're describing, we could make it so `labware.load` fn defaults its arg to `namespace=None`, and we could make that `None` mean "first look in 'opentrons' namespace, then if we're running a bundled protocol, search thru all the bundled defs for a single 'loadName' match and don't filter by namespace. If not exactly 1 of the bundled defs matches the loadName, throw an error." Version could work similarly when unspecified in bundled protocols.

(Also if the bundled labware get calibrated to `labwareName + "protocol" namespace + version`, my labware of the same version number with different namespaces will have competing calibration data across different protocol bundle runs)

3. Further off: if we can, I'd like to totally avoid users dealing with labware def files. I wonder if Run App could be responsible for downloading them during the bundling process. So if I did in a protocol 

```python
x = labware.load(load_name="something", namespace="my_lab", slot="1")
y= labware.load("opentrons-96-tiprack-300-ul", "2")
``` 

then the bundler would fetch the defs `my_lab/something` and `opentrons/opentrons-96-tiprack-300-ul` from the labware def cloud at bundling time and put them in the zip. Users never have to download a labware def file, unless they're doing Jupyter.

sfoster1, 2019-05-20:
1. `labware.load` would default to a permissive handling with a defined search order, starting with labwares bundled in the protocol and falling back to `custom` and then `opentrons` (with a message if it detects duplication).

2. I honestly forgot that protocols carry namespaces along with them, whoops. I'll change the RFC

3. That does seem like a laudable goal, though not for mvp. It requires either
  - (not great): parse the python protocol in the runapp ahead of time
  - (bit better): have a behind-the-scenes back and forth between the runapp and the api, where the api simulates, reports it's missing a labware, and the runapp goes and finds it and loads it in

I think you'd still want the explicit bundling at least as a fallback.

mcous, 2019-05-21:
As mentioned by @IanLondon in Slack, I think there's a big chunk of UX problems that come with labware bundling that mean we should punt on the labware bundling specifically. Arbitrary data bundling, however, definitely does not have those same problems.

Given that the RFC mentions a UI where the user selects multiple files to create a ZIP, I would like to pose this questions for discussion: is ZIP creation via the app necessary vs a multi-file upload? Maybe the following could work:

- Prereq - refactor protocol upload to occur over HTTP rather than RPC
- Option 1: User uploads Python file - place file in temp directory, proceed
- Option 2: User uploads ZIP file - unzip file into temp directory, proceed
- Option 3: User uploads multiple files - place all files in temp directory, find Python file, proceed

Moving to an HTTP system necessitates reconsidering how JSON protocols work at the upload stage, so we'd probably want to talk about that, too

sfoster1, 2019-05-22:
That does sound pretty tempting, and I think we wouldn't need to do anything special for json protocols - that's just another thing you can upload to the same endpoint.

However, taking upload over to http is actually a pretty major refactor because it changes the core flow of the rpc, which now manages protocol session creation directly. Internally we assume that a protocol session always has a protocol associated with it, since session.create takes the protocol text as a parameter. So this refactor requires either
- moving protocol session management out of rpc entirely, in favor of an http api that establishes some session resource you can establish a websocket to 
  - which breaks the app's "connect" concept
or
- having more complex interworkings between this new part of the http api and the rpc

Stefan1902, 2019-05-23:
A nice option for my use case would be a possibility to easy upload .csv files to the OT2, which can be accessed in the protocol.  A dropfield for the .csv files would be nice, that would save these files in data/csv or something like that.

btmorr, 2019-05-23:
We had a F2F conversation in the office today on this. What we arrived at:

RFC is approved

Work that we'll plan on doing relatively immediately ("immediately" specifically means by the time we ship an application that will provide a UI for creating custom labware definitions, which is on the roadmap for the upcoming quarter):

- Modify protocol upload endpoints in the server to accept zip files that fit the format described in this RFC (exactly one `*.py` file in the top level of the zip, any other files can be in the top level of the zip, and the zip may contain directories including valid Python modules)
- Determine a lookup sequence for where to find labware (in zip, fallback to database on robot)
- Add faculties to the server to handle pathing for finding other files in a bundle (e.g., so if a user includes `mydata.csv` in the top level, and includes `open('./mydata.csv')` in the protocol, it will work)
- Add data to labware calibration files to ensure that we are able to make sure that a prior calibration corresponds to a labware def in a file (e.g.: detect when the name is the same as another but the measurements are different, to avoid using an invalid calibration)
- Add UI elements for showing warnings to users (which will include warnings around labware from bundles)
- Modify "Open Protocol" dialog in Run App to allow users to select multiple files or a directory, and zip what is selected

Other notes that came up in the meeting:

- By doing this, we automatically enable users to write protocols that depend on other files, such as CSVs, their own Python libraries, etc
- Choosing to do this does not preclude us from developing other solutions later--for instance, we could also build a separate workflow, in which a labware definition could be sent to the robot independent of a protocol or bundle, and separate labware calibration from protocol run
- We also plan on building endpoints to allow a user to see what labware have been calibrated on the robot and when the most recent calibration occurred for each
- Consider modifying JSON protocols to be more similar (a zip containing one protocol) to reduce perceived difference between the two types of protocols
- Bundling will be a big benefit for publishing protocols to the future redesign of Protocol Library, and the redesign of that project will be smoother if we do this before we do the redesign

closed 2021-12-10 by mcous

mstone-modulus, 2022-04-26:
Hi,

Is there a formal specification of what a protocol bundle looks like? I've found references to bundling in the API docs (e.g. `ctx.bundled_data`) and this thread, but not specific details on the expected directory structure/contents.

Thanks!