Opentrons/opentrons #3494: Run App: Metadata fields uses the

Laura-Danielle, 2019-05-28:
## overview
In testing found that the App uses the name of the _file uploaded_ for Protocol Name rather than the key in metadata.

## current behavior
- If you change the "name" of the protocol in metadata nothing happens, only when you change file name does the Protocol Name change.

## expected behavior
- The metadata key should be used for the Protocol Name change, and fall back to file name if metadata cannot be found

Kadee80, 2019-06-06:
Just started looking into this issue. Looks like there were more changes than just the protocol name:

- `data.protocol-schema` => `data.schemaVersion` 
- `data.metadata.protocol-name` => `data.metadata.protocolName`
- `designer-application.application-name` => No longer present?

@IanLondon @Laura-Danielle are we planning on putting the `designer-application` info into the V3 protocol designer protocols? If not, we lose the creation method info for PD protocols on the file info page.

IanLondon, 2019-06-06:
@Kadee80 It's `designerApplication.name` where both `designerApplication` and `name` are optional keys

Kadee80, 2019-06-06:
@IanLondon same goes for `designer-application.application-version` => `designerApplication.version` ?

IanLondon, 2019-06-06:
@Kadee80 that's correct!

commit d2538382 (2019-06-10): feat(app): Get protocol name for all protocol schema versions

closes #3494
commit c15de443 (2019-06-11): feat(app): Get protocol name for all protocol schema versions

closes #3494
PR #3531 (2019-06-12): feat(app): Get protocolDisplayData based on protocol schema
## overview

This PR checks for protocol name, last modified, appName, and appVersion for based on protocol schema versions.

closes #3494

## changelog

- feat(app): Get protocol name for all V3 PD protocols

## review requests

Labeled this a WIP for now - this issue is a little larger than expected as it touches more than just the name field once we start getting schema version from shared data. It also uncovered the lack of v3 protocol file flow type.

[V2 protocol to test](https://github.com/Opentrons/opentrons/files/3262659/Kati [... cut]
commit ec69d849 (2019-06-12): feat(app): Get protocolDisplayData based on protocol schema (#3531)

closes #3494
closed 2019-06-12 by Kadee80