Opentrons/opentrons #3529: A protocol cannot be canceled when there are just module commands left

Cin316, 2019-06-06:
## overview

When just module commands are left in a protocol, even if the protocol is cancelled the module commands keep running.

## current behavior

In version 3.9.0, the magnet module keeps moving even after the protocol is cancelled.  I'm pretty sure this happens with the temperature module too, but I have not tested it extensively.

## steps to reproduce

1. Run the following protocol with the magnet module connected:

```
from opentrons import robot, labware, instruments, modules

magnet_module = modules.load('magdeck', '7')

plate = labware.load('trough-12row', 3)

tips = labware.load('opentrons-tiprack-300ul', '1')

pipette = instruments.P300_Single(mount ='right', tip_racks=[tips])

pipette.transfer(60, plate.wells('A1'), plate.wells('A2'))

for i in range(0, 45):
    magnet_module.engage(height=i)
    robot.comment('Engaged to height=' + str(i) + 'mm')
    pipette.delay(seconds=2)
```
2. After the pipette does the transfer and after the magnet module has started moving up, cancel the protocol in the Opentrons app.
3. The app will indicate that the protocol was cancelled.  However, the magnet module will keep moving up until the protocol completes.

## expected behavior

After a protocol is cancelled, no further module commands should be sent.  In the example above, the magnet module should stop moving.

mcous, 2019-11-19:
Potentially fixed in PAPIv2 but requires investigation to confirm

nusrat813, 2019-12-19:
**Issue persists in apiv1 and apiv2 for the magdeck** 

1. Protocol is canceled
2. Protocol pauses 
3. Robot homes 
4. Protocol resumes - magdeck continues increasing in engage height 
5. Reset run banner appears - once clicked robot still continues iterating through commands 
6. Protocol continues until run complete 
7. Run complete banner appears 

No issues with tempdeck when canceling during module commands

sfoster1, 2020-03-02:
Should be fixed by #4870

sfoster1, 2020-03-24:
Closed by #4871

closed 2020-03-24 by sfoster1