Opentrons/opentrons #3530: When a protocol is cancelled from a pause, the robot makes a sudden movement

Cin316, 2019-06-06:
## overview
When a protocol is cancelled from a pause, the robot suddenly starts moving towards its next command, but then stops and pauses.

## current behavior

When a protocol is cancelled from a pause, the robot starts performing its next command, but stops after a fraction of a second.  This behavior is counterintuitive and could be dangerous, as a user might pause a protocol and then cancel it if something goes wrong. 
 Having the robot suddenly move on a cancel would be unexpected.

It appears this issue is connected with the fix done in #1443.

## steps to reproduce

1. Upload and run the following protocol:
```
from opentrons import robot, labware, instruments

source_rack = labware.load('96-PCR-flat', '5')
dest_rack = labware.load('96-PCR-flat', '2')

tiprack = labware.load('opentrons-tiprack-300ul', '4')

pipette_left = instruments.P300_Single(mount ='right', tip_racks=[tiprack])

pipette_left.transfer(25, source_rack.wells('A2'), dest_rack.wells('A2'))
robot.pause()
pipette_left.transfer(25, source_rack.wells('A3'), dest_rack.wells('A3'))
```
2. When the robot pauses between the 2 transfers, cancel the protocol.
3. The robot will suddenly start moving in the direction of its next command.

## expected behavior

When a protocol is cancelled (whether from a pause or not), all movement of the robot should stop immediately (or remain stopped if the robot was already paused).

mcous, 2019-06-25:
Thanks for the report @Cin316. You're right, this is a known behavior caused by #1443. That change was introduced due to fundamental architecture problems with the protocol run API. Our options were basically:

1. Protocol cancel doesn't always work if paused
2. Protocol cancel works if paused, but we need to allow a "resume" first that may move the robot

This behavior will remain in place until we are able to rewrite this part of the robot software. This project is in its planning stages, and I'll be sure to include / update this ticket in those plans

amitlissack, 2021-01-29:
I'm pretty sure that ExecutionManager resolved this issue.

closed 2022-03-04 by mcous