Opentrons/opentrons #3535: rpc: Add a warning message type

sfoster1, 2019-06-06:
To enable us to display warning messages sourced from the api without halting execution for errors, we should add a warning essage to the rpc. This message should be structured like
```
{
  "$": {
     "type": WARNING_TYPE # This is a new constant to add
  },
  "message": str, # a human-readable warning message
  "origin": str, # a human-readable string indicating the source of the warning (i.e. protocol line)
}
```

Add this in a couple places where we today raise `RuntimeWarning` or use the `warnings` module.

mcous, 2019-06-25:
If at all possible, I would really like to avoid sinking any more feature work into RPC. What are the specific warnings we're trying to add, and could they be surfaced in the simulation stage?

SyntaxColoring, 2021-07-01:
Closing this as `wontfix` in favor of #8042. Our RPC subsystem will soon be deprecated, but we still need to address this in the HTTP API that's replacing it.

closed 2021-07-01 by SyntaxColoring