Opentrons/opentrons #3574: Pipette Gen2: P1000 config modifications

sfoster1, 2019-06-11:
HWeng indicates the following changes to p1000 config:
- Modify aspirate/dispense function
- Add pickup tip shake by default

ChrisYarka, 2019-06-26:
Function Update:

50.05 | 0.0456 | 23.83
77.11 | 0.0250 | 23.78
104.6 | 0.0164 | 24.44
132.1 | 0.0095 | 25.16
187.3 | 0.0062 | 25.60
215.0 | 0.0041 | 25.99
325.6 | 0.0024 | 26.36
436.1 | 0.0011 | 26.76
1075 | 0.0004 | 27.07

[image]

ChrisYarka, 2019-06-26:
steps/mm = 3200

ChrisYarka, 2019-06-26:
P1000 Plunger Positions:
top = 19.5 
bottom = -18.5
blowout = -23
drop tip = -37

ChrisYarka, 2019-06-26:
[image]

ChrisYarka, 2019-06-26:
https://docs.google.com/spreadsheets/d/1faebzB-o9H1EFUyCtUBaGqAWodQFq5mfhqbgBL8L8v4/edit#gid=212246552

PR #3690 (2019-07-09): refactor(api):  Modify Gen2 single channel configs from hw testing
## overview
Closes #3576, closes #3575, closes #3574. 

## changelog
- Update requested configs from tickets above
- Modify steps per mm value to 3200

## review requests

Double check all changes made it into this PR
closed 2019-07-09 by Laura-Danielle