Opentrons/opentrons #3575: Pipette Gen2: P20 Config Modification

sfoster1, 2019-06-11:
For DVT, the p20 needs the following config changes:

- New aspirate/dispense function
- New plunger positions

ChrisYarka, 2019-06-26:
P20 Aspirate AND Dispense Function:

1.307 | 0.0666 | 0.6237
2.095 | 0.0566 | 0.6428
2.865 | 0.0233 | 0.6675
3.600 | 0.0211 | 0.7054
4.412 | 0.0190 | 0.7224
5.135 | 0.0120 | 0.7448
5.900 | 0.0051 | 0.7111
8.913 | 0.0017 | 0.7325
12.01 | 0.0015 | 0.7276
15.11 | 0.0014 | 0.7379
18.22 | 0.0013 | 0.7402
22.00 | 0.0011 | 0.7423

[image]

ChrisYarka, 2019-06-26:
steps/mm = 3200

max speed = 30 mm/s

quirks = no change. keep the drop tip shake.

ChrisYarka, 2019-06-26:
Plunger Positions:

top = 19.5
bottom = -8.5
blowout = -13
dropTip = -27

ChrisYarka, 2019-06-26:
[image]

ChrisYarka, 2019-06-26:
https://docs.google.com/spreadsheets/d/1faebzB-o9H1EFUyCtUBaGqAWodQFq5mfhqbgBL8L8v4/edit#gid=442591547

PR #3690 (2019-07-09): refactor(api):  Modify Gen2 single channel configs from hw testing
## overview
Closes #3576, closes #3575, closes #3574. 

## changelog
- Update requested configs from tickets above
- Modify steps per mm value to 3200

## review requests

Double check all changes made it into this PR
closed 2019-07-09 by Laura-Danielle