Opentrons/opentrons #3576: Pipette Gen2: P300S config modifications

sfoster1, 2019-06-11:
Pipette Gen2 P300s needs the following config modifications:

- Plunger positions

ChrisYarka, 2019-06-26:
P300 Function Update

17.060 | 0.0472 | 8.1073
26.239 | 0.0236 | 8.1277
35.458 | 0.0128 | 8.4095
44.745 | 0.0091 | 8.5417
63.509 | 0.0070 | 8.6355
110.426 | 0.0062 | 8.6574
148.131 | 0.0038 | 8.7878
330.000 | 0.0005 | 9.1770

[image]

ChrisYarka, 2019-06-26:
steps/mm = 3200

max speed = 30 mm/s

quirks = no change from EVT. keep the drop tip shake.

ChrisYarka, 2019-06-26:
P300 Plunger positions

top = 19.5
bottom = -14.5
blowout = -19
dropTip = -37

ChrisYarka, 2019-06-26:
[image]

ChrisYarka, 2019-06-26:
https://docs.google.com/spreadsheets/d/1faebzB-o9H1EFUyCtUBaGqAWodQFq5mfhqbgBL8L8v4/edit#gid=1540046538

PR #3690 (2019-07-09): refactor(api):  Modify Gen2 single channel configs from hw testing
## overview
Closes #3576, closes #3575, closes #3574. 

## changelog
- Update requested configs from tickets above
- Modify steps per mm value to 3200

## review requests

Double check all changes made it into this PR
closed 2019-07-09 by Laura-Danielle