Opentrons/opentrons #3582: Thermocycler: Open and Close Lid With HW Button

umbhau, 2019-06-12:
PLACEHOLDER; initial thoughts below. TBD clarifying roles of software and hardware buttons.

As a Thermocycler user, I would like to be able to open or close the Thermocycler using the physical button on the module.

## Acceptance Criteria
- [ ] Pushing button toggles physical lid state (opens if closed; closes if open)
- [ ] If temp is above 70C, do not open lid, and alert user in Run App that the lid can only be opened when the temperature is below 70C
- [ ] If Thermocycler is active (heating, cooling, maintaining a target temp), do not open lid, and alert user in Run App that the protocol must be paused before opening the lid
- [ ] User should receive visual feedback if they are prevented from opening the lid, e.g. flashing the light 3 times rapidly.
- [ ] Allow user to open lid in all scenarios (overriding above constraints) by holding down button for for five seconds 
- Note user will get an alert in the Run App that lid is opened, per #3134

pantslakz, 2019-08-01:
Updates to this flow:

- user able to open/close regardless of temperature
- user able to open/close mid cycle without disrupting run
- only run pause or run stop enables user option to "deactivate" TC

bndo, 2019-08-06:
This is addressed in #3808

closed 2019-08-06 by bndo