Opentrons/opentrons #3669: Unexpected Movement after Run Cancel

umbhau, 2019-07-02:
## Expected Behavior
- After cancelling a run, robot should stop all movement. Robot should home z axis.

## Observed Behavior
- Robot continues moving after cancel is confirmed. 
- Note observed movement sometimes reflects next planned movement in protocol, and sometimes appears random

Note there is sometimes a delay after user clicks 'cancel' and before robot stops; this is okay. This bug refers to additional movement after user clicks 'confirm' in the cancel dialogue box.

## Steps to Reproduce
- Unable to reproduce this 100% of the time or find a clear pattern, but it does happen fairly often (25% of the time?)
- In general: 
  1. Start a protocol
  2. Click 'cancel'
  3. Click 'confirm'
  4. Observe robot make additional movements

kinnari-opentrons, 2019-07-08:
[protocol_for_cancel_bug.txt](https://github.com/Opentrons/opentrons/files/3370229/protocol_for_cancel_bug.txt)

ChrisYarka, 2019-07-08:
@umbhau 

Steps to reproduce this issue:
- Run the protocol Kinnari attached. Requires mag and temp decks.
- Hit cancel during the run. 

After cancelling, the robot will continue to perform additional actions. Some times the actions are normal but sometimes they will crash into seemingly random places.

sfoster1, 2020-02-03:
Duplicate of #3530

closed 2020-02-03 by sfoster1