Opentrons/opentrons #3676: PD: Multichannel pipettes crash when dispensing to trash

opatel0, 2019-07-08:
## overview

## current behavior

## steps to reproduce

## expected behavior

opatel0, 2019-07-08:
Here are the customer protocols 
[drug addition 2.zip](https://github.com/Opentrons/opentrons/files/3370144/drug.addition.2.zip)
[adding drug3.zip](https://github.com/Opentrons/opentrons/files/3370166/adding.drug3.zip)

IanLondon, 2019-07-08:
Confirmed that on edge d214c5adef15d4b575beaf29584caa375b80d4f3 v3 JSON protocols still have this issue.

IanLondon, 2019-07-09:
Also in APIv1 Python protocols, dispense to trash eg `multichannel.transfer(8, samples['A1'], robot.fixed_trash['A1'])` will cause an identical crash.

IanLondon, 2019-09-03:
Blocked - requires a significant amount of APIv1 refactoring. Hopefully we can move to APIv2?

umbhau, 2019-09-03:
Closing. Will not fix in API v1; fixed in API v2.

btmorr, 2019-09-10:
Changing APIv1 server to use APIv2 labware definitions allows us to fix this in an acceptable way. PR forthcoming

btmorr, 2019-09-10:
Oh, nevermind. I thought an in flight change was going to fix this, but this is intractable.

closed 2019-09-10 by btmorr