Opentrons/opentrons #3738: Pipetting Behavior: Drop Tips After Cancel to Avoid Contamination

umbhau, 2019-07-17:
As an OT-2 user, I would like my OT-2 to drop tips after I cancel a protocol, so that my pipettes aren't contaminated when the OT-2 homes its plunger axis.

## Acceptance Criteria
- [ ] Drop tips in fixed trash after user cancels a run, before homing plunger axis

btmorr, 2019-08-01:
Should this be "drop tip before homing plunger"?

James-Kitson, 2019-08-16:
I'd like to second this, I had to cancel a run today and I forgot that the robot does this. It managed to suck liquidised tissue through a filter up into the pipette before I spotted what was happening.

Ideal behaviour:
1. Cancel run.
2. Pipette moves to fixed waste bin and drops tips without any aspirating action.
3. Pipette does any homing that is required.

umbhau, 2019-08-16:
@btmorr Yes, updated the issue -- now generalized to any time the pipette homes its plunger axis.

@James-Kitson Thank you for the comment -- this one is on our radar and near the top of the priority list.

SyntaxColoring, 2020-04-14:
Changed the title of this ticket to match the August 16, 2019 edit, which I think mistakenly just edited the body.

SyntaxColoring, 2020-04-14:
I'm not 100% sure, but I think ejecting tips before homing plungers isn't always going to be possible.

In particular, when the OT-2 boots, it won't know the pipettes' plunger positions until they're homed for the first time. So, it won't know how far to move the pipette motor to engage the tip ejector.

PR #8229 (2021-08-30): fix(api): drop tip after cancel
Closes #7466
Closes #3738
Closes #5428

# Overview

This PR adds a functionality that if a protocol is cancelled mid-run and the pipette has a tip attached when the cancel is initiated, then the robot will empty the tip & drop it in the pipette's trash container before homing the plungers.

This is achieved in the following way:

Right after a cancel is issued:
1. pipette state is stored to determine whether it has a tip (and to save the last known plunger location but I’m not using that right now)
2. Cancel is fully executed with `home_after=False`. Hardware co [... cut]
closed 2021-08-30 by sanni-t