Opentrons/opentrons #3757: USA Scientific deep well plate is incorrectly marked as magnetic module ok

sfoster1, 2019-07-19:
## current behavior

`usascientific_96_wellplate_2.4ml_deep` is marked as magnetic module compatible and the system would allow you to use it in an apiv2 protocol. It is not explicitly listed as supported in the apiv1 magnetic module driver, however.

https://github.com/Opentrons/opentrons/blob/08e25e20edc5ddcd319bf037f57ee86dd30b43b7/shared-data/labware/definitions/2/usascientific_96_wellplate_2.4ml_deep/1.json#L1104-L1105

## expected behavior

Either remove the capability from the v2 labware (probably by uploading a new version) or add it to the compatibility list for the magnetic module in v1.

Laura-Danielle, 2019-07-20:
So we _do_ have customers using this labware on the magnetic module, but to accurately use it for science purposes you need to have an adapter (so the magnets in the home position don't also pull the beads).

This might be an @umbhau question.

mcous, 2019-08-29:
Gonna close this because it seems like this definition is actually correct. Can re-open if we decide to change this

closed 2019-08-29 by mcous