Opentrons/opentrons #3766: pass a parameter from local python script to the function in the script in Opentrons

ffffffrank, 2019-07-22:
## overview

## current behavior

## steps to reproduce

I'm expecting the i passed from local, would be in dist_sample(i). For example, if i is 0 in local, when I call stdin,stdout,stderr=ssh_client.exec_command('bash -lc "python3 /data/test.py dist_sample(i)"'), the dist_sample should also take parameter i as 0.

mcous, 2019-07-29:
Hi @ffffffrank! A few things here:

- `python3 /some/file.py function_name()` is **not** a valid way to call a function from the command line
- In your local code sample, it looks like your sending the literal character `"i"` instead of inserting the value via `.format` or similar

When you execute a python script from the command line, all values after the script file name get passed to the program as string command line arguments. I would recommend researching how to use the [argparse](https://docs.python.org/3.6/library/argparse.html#module-argparse) module to read CLI arguments on your remote script.

You'll also need to modify your local script to pass the value of `i` into the call:

I don't know if this will work, but hopefully this could be a jumping off point for you to get something working:

```py
# local file
for i in range(25):
    stdin, stdout, stderr = ssh_client.exec_command(
        f'bash -lc "python3 /data/test.py {i}"'
    )

# remote file
import argparse

parser = argparse.ArgumentParser(description="Distribute liquids")
parser.add_argument("source", metavar="S", type=int, help="Source well")

args = parser.parse_args()
dist_sample(args.source)

def dist_sample(i):
    pipette.pick_up_tip()
    pipette.distribute(400, sample.wells(i), p12.rows("A"), new_tip="never")
    pipette.drop_tip()
```

I'm going to close this seeing as it's a support request, but feel free to continue using this thread for questions if you're still having trouble getting this to work

closed 2019-07-29 by mcous

ffffffrank, 2019-07-29:
> Hi @ffffffrank! A few things here:
> 
> * `python3 /some/file.py function_name()` is **not** a valid way to call a function from the command line
> * In your local code sample, it looks like your sending the literal character `"i"` instead of inserting the value via `.format` or similar
> 
> When you execute a python script from the command line, all values after the script file name get passed to the program as string command line arguments. I would recommend researching how to use the [argparse](https://docs.python.org/3.6/library/argparse.html#module-argparse) module to read CLI arguments on your remote script.
> 
> You'll also need to modify your local script to pass the value of `i` into the call:
> 
> I don't know if this will work, but hopefully this could be a jumping off point for you to get something working:
> 
> ```python
> # local file
> for i in range(25):
>     stdin, stdout, stderr = ssh_client.exec_command(
>         f'bash -lc "python3 /data/test.py {i}"'
>     )
> 
> # remote file
> import argparse
> 
> parser = argparse.ArgumentParser(description="Distribute liquids")
> parser.add_argument("source", metavar="S", type=int, help="Source well")
> 
> 
> args = parser.parse_args()
> dist_sample(args.source)
> 
> 
> def dist_sample(i):
>     pipette.pick_up_tip()
>     pipette.distribute(400, sample.wells(i), p12.rows("A"), new_tip="never")
>     pipette.drop_tip()
> ```
> 
> I'm going to close this issue since it's a support request, but feel free to continue using this thread for questions if you're still having trouble getting this to work

Hi Mike,

Thanks for your help.

just a quick question. So if I don't need to pass any value to the remote script. Can I just call the function directly in local like this?
stdin, stdout, stderr = ssh_client.exec_command('bash -lc "python3 /data/test.py dist_sample()"' )

mcous, 2019-07-29:
> Can I just call the function directly in local like this?
stdin, stdout, stderr = ssh_client.exec_command('bash -lc "python3 /data/test.py dist_sample()"' )

No, unfortunately that's not how executing Python from the command line works. When you call `python` (or `python3`) on a file, you're doing `python file_name.py args...`. Every space-separated thing in `args...` is simply treated as a string. Python doesn't know (or care) that you've written a function name.

If you'd like to execute dist_sample remotely, I'd recommend one of these options depending on your needs:

- Add a call to `dist_sample` in your remote Python file to execute if the script file is called directly
    - This will work if you don't ever need to call other methods in the file
    - Local:
    ```py
    stdin, stdout, stderr = ssh_client.exec_command('bash -lc "python3 /data/test.py"')
    ```
    - Remote:
    ```py
    def dist_sample():
        pipette.pick_up_tip()
        pipette.distribute(400, sample.wells(), p12.rows("A"), new_tip="never")
        pipette.drop_tip()

    if __name__ == "__main__":
        dist_sample()
    ```
- If you need to call other methods, configure `argparse` to expect a method name as the first argument, e.g. `python3 my_file.py method_name arg1 arg2 arg3`
    - This could get a little complicated depending on the needs of your other methods
    - I would recommend generally researching Python command-line interfaces and argparse
    - That should be more helpful than I am able to be

Depending on how your import paths, you may also want to look into `python -c "..."`, but I'm not sure how to get that to work with an absolute path to a script remotely, which is why I recommended the options above

ffffffrank, 2019-07-30:
> > Can I just call the function directly in local like this?
> > stdin, stdout, stderr = ssh_client.exec_command('bash -lc "python3 /data/test.py dist_sample()"' )
> 
> No, unfortunately that's not how executing Python from the command line works. When you call `python` (or `python3`) on a file, you're doing `python file_name.py args...`. Every space-separated thing in `args...` is simply treated as a string. Python doesn't know (or care) that you've written a function name.
> 
> If you'd like to execute dist_sample remotely, I'd recommend one of these options depending on your needs:
> 
> * Add a call to `dist_sample` in your remote Python file to execute if the script file is called directly
>   
>   * This will work if you don't ever need to call other methods in the file
>   * Local:
>   
>   ```python
>   stdin, stdout, stderr = ssh_client.exec_command('bash -lc "python3 /data/test.py"')
>   ```
>   
>   
>   
>   * Remote:
>   
>   ```python
>   def dist_sample():
>       pipette.pick_up_tip()
>       pipette.distribute(400, sample.wells(), p12.rows("A"), new_tip="never")
>       pipette.drop_tip()
>   
>   if __name__ == "__main__":
>       dist_sample()
>   ```
> * If you need to call other methods, configure `argparse` to expect a method name as the first argument, e.g. `python3 my_file.py method_name arg1 arg2 arg3`
>   
>   * This could get a little complicated depending on the needs of your other methods
>   * I would recommend generally researching Python command-line interfaces and argparse
>   * That should be more helpful than I am able to be
> 
> Depending on how your import paths are set up, you may also want to look into `python -c "..."`, but I'm not sure how to get that to work with an absolute path to a script remotely, which is why I recommended the options above

Got it, thanks for your help!