Opentrons/opentrons #3767: Thermocycler: Alert User of Module Malfunction

umbhau, 2019-07-22:
As a Thermocycler user, I would like to know if my Thermocycler experiences an internal (firmware) failure or malfunction.

## Acceptance Criteria
- [ ] Alert user in Run App (with a dismissible modal) if TC module returns a failure message.

## Content
- TBD

## Design
- TBD

btmorr, 2019-07-25:
Should lid auto-open? How do we envision the user recovering from this?

btmorr, 2019-08-29:
cc @pantslakz for designs re: how do we show the error to users

btmorr, 2019-09-26:
This is now centered around physical failures in the Thermocycler, such as a thermistor failure. In these cases, we will raise an error that will terminate a protocol run, and use the existing error modals to show an error message to the user with actionable info.

b-cooper, 2019-12-09:
This is addressed in a firmware change in https://github.com/Opentrons/opentrons-modules/pull/96 .  The module itself now has a visible error state.  This should eventually be accompanied by a runtime message, though that messaging should ideally be included once the current RPC session api implementation has been replaced.

closed 2019-12-09 by b-cooper