Opentrons/opentrons #3796: Labware Definition: NEST 96 Well Plate 100 µL PCR Full Skirt

umbhau, 2019-07-29:
As an Opentrons user, I would like to be able to use NEST PCR plates on my OT-2.

## Acceptance Criteria
- [ ] Create a labware def for the following labware:
  - Display Name:	NEST 96 Well Plate 100 µL PCR Full Skirt
  - API Name: nest_96_wellplate_100ul_pcr_full_skirt
- [ ] Labware definition should match manufacturer's technical diagram
- [ ] Labware should be browseable with images in Labware Library. 

## Resources
- Labware information in the 'Labware Backlog' tab of the '[Labware Definitions](https://docs.google.com/spreadsheets/d/1hXi63baR-oBM9HwKvWDP_Aba7VWVz1AA4dn9kixXgxA/edit#gid=2128909968
)' sheet.
- Labware technical diagram here: https://drive.google.com/drive/u/0/folders/1EJGJAV8y1d6G_v9vTakb09CecYWmMULM

- Please use images from non filter tip racks, located here: https://drive.google.com/drive/u/0/folders/1eyTLAi7XtEfuwPdTWq_2eNhN6c10ulME

umbhau, 2019-07-29:
Blocked pending more detailed labware diagram

ahiuchingau, 2019-08-05:
- Updated Labware technical drawing here:
  https://drive.google.com/file/d/1ym4rSuzUsr5jtR70P-oELeMu-zU8kK7r/view