Opentrons/opentrons #3809: Labware v2: Bug calibrating multiple "opentrons_96_tiprack_10ul" tipracks

hainesm6, 2019-08-01:
## overview

Bug when using multiple "opentrons_96_tiprack_10ul" tipracks.

## current behavior

When calibrating 5 "opentrons_96_tiprack_10ul" tipracks in deck positions "['3', '6', '9', '2', '5']", calibration is offset and tips are not correctly picked up.

## steps to reproduce

from opentrons import labware, instruments, robot # robot class added.

CANDIDATE_TIPRACK_SLOTS = ['3', '6', '9', '2', '5']
TIPRACK_TYPE = 'opentrons_96_tiprack_10ul'
PIPETTE_MOUNT = 'right'
DEFAULT_HEAD_SPEED = {'x': 600, 'y': 400, 'z': 125, 'a': 125}
SLOW_HEAD_SPEED = {'x': 600//10, 'y': 400//10, 'z': 125//10, 'a': 125//10}
tipracks = [labware.load(TIPRACK_TYPE, slot)
            for slot in CANDIDATE_TIPRACK_SLOTS]
pipette = instruments.P10_Single(mount=PIPETTE_MOUNT, tip_racks=tipracks)
for tiprack in tipracks:
    for tip in (tiprack[0], tiprack[-1]):
        robot.head_speed(combined_speed=max(SLOW_HEAD_SPEED.values()),
                         **SLOW_HEAD_SPEED)
        pipette.pick_up_tip(tip)
        robot.head_speed(combined_speed=max(DEFAULT_HEAD_SPEED.values()),
                         **DEFAULT_HEAD_SPEED)
        pipette.drop_tip()

## expected behavior

Should function correctly as it does when TIPRACK_TYPE = 'tiprack-10ul'.

sfoster1, 2019-08-01:
@hainesm6 What version of the Opentrons app and robot software are you on? You can view your Opentrons App version by clicking the three dots at the bottom left, and check the robot software version by clicking on your robot in the left pane.

I ask because this sounds like an issue that we fixed in version 3.10.3.

hainesm6, 2019-08-01:
@sfoster1 I've checked both and they say "3.10.3".

sfoster1, 2019-08-01:
Is the calibration offset by the same amount, or does it vary between the different tipracks across the deck?

hainesm6, 2019-08-01:
Yep, it's offset by the same amount across all tipracks.

sfoster1, 2019-08-01:
We'll look into it internally, but for the immediate issue please contact our support team using the button on the bottom right of the app, or via email at support@opentrons.com.

hainesm6, 2019-08-02:
Thank you @sfoster1. I've spoken to support and until it's fixed will define tipracks as "tiprack-10ul".

bndo, 2019-09-03:
This is being addressed via #3938

closed 2019-09-03 by bndo