Opentrons/opentrons #3885: Labware: Opentrons 96 Well Tube Rack with NEST 0.5 mL Screwcap

umbhau, 2019-08-19:
As an Opentrons user, I would like to be able to use NEST PCR plates on my OT-2.

## Acceptance Criteria
- [ ] Create a labware def for the following labware:
  - Display Name: Opentrons 96 Well Tube Rack with NEST 0.5 mL Screwcap
  - API Name: opentrons_96_tuberack_nest_0.5ml_screwcap
- [ ] Labware definition should match manufacturer's technical diagram
- [ ] Labware should be browseable with images in Labware Library. 

## Resources
- Labware information in the 'Labware Backlog' tab of the '[Labware Definitions](https://docs.google.com/spreadsheets/d/1hXi63baR-oBM9HwKvWDP_Aba7VWVz1AA4dn9kixXgxA/edit#gid=2128909968
)' sheet.
- Labware technical diagram here: https://drive.google.com/drive/u/0/folders/1EJGJAV8y1d6G_v9vTakb09CecYWmMULM

- Images
https://drive.google.com/drive/u/0/folders/1eyTLAi7XtEfuwPdTWq_2eNhN6c10ulME

umbhau, 2019-08-20:
closing, we don't have a 96 well tube rack holder.

ahiuchingau, 2019-08-21:
Reopening this issue as the 0.5 mL screwcap tubes actually would fit in the 24 well tube rack