Opentrons/opentrons #3899: PL: Provide deck data from the executor

Laura-Danielle, 2019-08-20:
## overview
We need some way to provide data to the protocol library about the deck setup. This is currently done by having the protocol library first simulate the protocol to check for syntax errors and then parse the file for labware/module/pipette info.

## expected behavior
Instead, the executor method from the API should also provide the deck information during simulation. @mcous could the app also eventually use this information instead of relying on the RPC?

IanLondon, 2019-09-13:
Labware & pipettes parsed out in `apiv2` branch. However, in PL the pipettes never show up. Modules TBD, not done yet, not sure how to parse out.

IanLondon, 2019-09-16:
Closed in `apiv2` branch. Making pipettes show up is not an existing PL feature and needs design, will be separate ticket if we do it.

closed 2019-09-16 by IanLondon