Opentrons/opentrons #3909: PL: Vue correctly handles new parameters format

Laura-Danielle, 2019-08-20:
## overview

Once the parser correctly injects a parameters section in JSON for each protocol, vue needs to be able to interpret the new format correctly.

## expected behavior

The json blob is in the form of:

```
{
    ‘pipettes’: {
        ‘Type’: ‘dropDown’
        ‘default’: ‘p300_single’
        ‘options’: [‘p50_single’, ‘p300_single’, ‘p20_single’],
        ‘value’: ‘p300_single’
     },
  ‘cherrypickWells: {
         ‘Type’: ‘file’
         ‘default’: ‘’, ..etc
}
```
a. The top level key will be the name used by the UI for the parameter name.
b. The `type` key should notify vue what other options to expect within the top level key (i.e. if type is `dropdown`, there should be a list of options for that dropdown.

IanLondon, 2019-09-11:
**Proposed changes**

Data type (in Flow notation)

```js
Array<{|
    type: "dropDown",
    label: string,
    options: Array<{|
        name: string,
        value: string
    |}>,
    // for dropDown, default is the first option in the array
|} | {|
    type: "textFile",
    label: string,
    default: string
|}>
```

Example:

```json
[
    {
        "type": "dropDown",
        "label": "Pipette",
        "options": [
            {
                "name": "P10 Single-Channel",
                "value": "P10_Single"
            },
            {
                "name": "P10 Multi-Channel",
                "value": "P10_Multi"
            }
        ],
        "default": "P10_Single"
    },
    {
        "type": "textFile",
        "label": "Some CSV",
        "default": "1,2,3,4\n5,6,7,8"
    }
]
```

* Field objects are ordered, the order determines how they'll display on the page
* `label` is the text labelling the field on the page
* 2 `type`s (just like there's 2 `otcustomizers` things in PL right now) -- dropDown and textFile

IanLondon, 2019-09-16:
Closed in `apiv2` branch

closed 2019-09-16 by IanLondon