Opentrons/opentrons #3928: App: UI for installing SSH keys

sfoster1, 2019-08-22:
## overview

Release 3.11.0 restricted SSH access to the OT-2 to public-key authentication. This changes users workflow and requires a more elaborate setup to allow remote access to the robot, as detailed in this intercom article: https://app.intercom.io/a/apps/bsgvg3q7/articles/articles/3203681/show

We should add a UI for people to at least upload, and perhaps manage, public keys on their robot. 

## API endpoints

This uses the following HTTP API, which is available *only* when connected via USB:

### Add a key
`POST /server/ssh_keys`, body application/json: `{"key": "key string"}`

The key string is the contents of the public key, in openssh format only. The host identifier at the end may be excluded. The algorithm at the beginning must be present. Only the `ssh-rsa` algorithm is supported, not `ecdsa`. Note that keys generated with e.g. PuTTY will _not_ be in this format, and must be converted.

Response codes:
`201 Created`, body `{"message": "added key {hashval}", "key_md5": "hashval"}`
The hash is the md5 hash of the body of they key.

`400 Bad Request`, body: `{"error": "bad-key" | "no-key", "message": some error message}`
If the key is in the wrong format, has an unsupported algorithm, has bad newlines, doesn't exist, etc.

### List Present Keys

`GET /server/ssh_keys`

List the keys installed on the robot. Response shape:
`200 OK`, body: `{"public_keys": [{"key_md5": digest, "key": key string},...]`

### Delete Key

`DELETE /server/ssh_keys/{key_hash}`

Delete a key by its known hash. This should use a hash returned by `GET /server/ssh_keys`.

Responses:

`200 OK` if the key was found and deleted
`404 Not Found` if the hash was not valid

Laura-Danielle, 2019-08-22:
We might also need a support article explaining how to set up ssh keys. There are probably a few users who don't know how to do that, but who may want to ssh in.

sfoster1, 2019-08-22:
That's the intercom article linked above

SyntaxColoring, 2021-06-03:
Deprioritizing this because, currently:

* For more savvy users, the documentation that guides them through [POSTing an HTTP endpoint](https://support.opentrons.com/en/articles/3203681-setting-up-ssh-access-to-your-ot-2) is usable enough.
* For less savvy users, Support currently relies on Jupyter Notebook, which has its own terminal. (Yes, this is a security problem that we should eventually deal with, at which point we should also revisit this ticket.)

closed 2025-12-15 by sfoster1