Opentrons/opentrons #3967: API v2: Cannot load pipette with name, only model

Laura-Danielle, 2019-08-29:
## overview
There is a regression in API v2 where you can no longer load in a pipette by its colloquial name (i.e. `p20_single_gen2`), but instead you need to pass in the entire model string (i.e. `p20_single_v2.0`).

## expected behavior

You should be able to pass in both a model string and a regular pipette name without behavioral differences.

ahiuchingau, 2019-09-10:
Tested the following protocol on `edge` and could not reproduce the above bug:
```
def run(ctx):
    from opentrons.types import Location, Point
    tiprack = ctx.load_labware('opentrons_96_tiprack_20uL', '6')
    pip = ctx.load_instrument('p20_single_gen2',
                              mount='right', tip_racks=[tiprack])
   
    pip.pick_up_tip()
    location = Location(Point(380.87, 9.0, 0.0), ctx.deck)
    pip.move_to(location)
    pip.return_tip()
```
Keeping this issue open until APIv2 release to remind us to test and make sure this no longer exists.

btmorr, 2019-12-03:
Not sure when this got fixed, but it's no longer the case

closed 2019-12-03 by btmorr