Opentrons/opentrons #3971: LC: upon file upload, show form & scroll

IanLondon, 2019-08-29:
## BLOCKED

Blocked by prereq, #3966 

Blocked by, or interferes with, scroll behavior of #3965 

## overview

When you upload a labware to LC, it kinda seems like nothing happened.

If a user uploads a well plate or reservoir, they should get feedback that the upload happened, and be encouraged to review the entire form.

If a user uploads a tube rack or aluminum block, LC cannot infer the tube rack child type / aluminum block child type fields. LC needs the user to fill in those fields before continuing.

## acceptance criteria

If you upload a labware (doesn't matter if it's via clicking the upload button, or via drag-and-drop the file), then...
- [ ] If the upload fails (bad file, unsupported labware) then you just get the existing error modal behavior and aside from that, nothing happens. No scroll, no content hide/show.
- [ ] If it's a well plate or reservoir, the page scrolls to the top of the form contents, and the form contents become visible appear (same exact effect as clicking the "Start Creating Labware" button)
- [ ] If it requires the user to fill in labware type child fields (for tube rack or aluminum block), form contents should be hidden and the scroll should not happen

pantslakz, 2019-09-03:
@IanLondon one outstanding concern - if user is able to swap child category during upload - then maybe we should consider resetting the api load name/display name to default?

pantslakz, 2019-09-04:
Hey Ian - some feedback on http://sandbox.labware.opentrons.com/lc_post-upload-behavior-3971/create:

- when user uploads labware which is aluminum block/tuberack, it should not load child in both "create new" and "upload labware" section.
- instead the 'create new' section should become inactive and not mimic action in "upload" section
- only the "upload" section should show parent
- If user wants to access 'create new' - I assume they would reload the page

Another behaviour thing which maybe best tracking in another ticket:
- if user exports at 96 well tube rack, and user re-uploads it in the LC and picks "24" tube as child -the suggested display/api name of labware does not change and remains 96.  This feels like further reasoning to re-set suggested name each time a definition is re-uploaded.

PR #4020 (2019-09-10): feat(labware-creator): separate create vs upload fields; scroll upon upload
## overview

Closes #3971

## changelog

## review requests

- [ ] uploading a file disables the "Create New Definition" section & hides its fields
- [ ] for wellPlate/reservoir, uploading reveals & scrolls down to the form content right away
- [ ] for aluminum block / tube rack, filling out the remaining fields and clicking "start editing labware" reveals & scrolls down to the form content
commit 5d63ce1b (2019-09-10): feat(labware-creator): separate create vs upload fields; scroll upon upload (#4020)

closes #3971
closed 2019-09-10 by IanLondon