Opentrons/opentrons #3985: Provide a way to do tip probing / pipette calibration that does not rely on a pre-existing good calibration

SyntaxColoring, 2019-08-30:
# Background

To find the length of a pipette tip, we move vertically down until we hit the z-switch in the box. To find the (x, y) of that z-switch, we take the average of the -x, +x, -y and +y switches positioned around it. To find the (x, y)s of *those* switches, we use the pre-existing calibration and hope it's good enough.

So, pipette calibration will be unreliable, sometimes crashing the tip, if it wasn't already basically correct to begin with.

For most cases, this apparently works. But for CLI calibration, at least, we want something more robust. Calibration tools should be able to cope with especially bad calibrations.

This was a real issue with my robot. It kept missing the switches during the tip probe stage of CLI calibration after giving it a fresh SD card. We only managed to bootstrap it by assisting pipette calibration with our fingers and repeating, until it was good enough to do the whole thing on its own.

# Proposal

During CLI calibration, have the user manually jog the pipette tip in front of the west (-x) switch. From that user-provided known-good starting position, we're guaranteed to be able to find the other three switches. (As long as #3983 is fixed!)

## Variants

  - Instead, manually jog to the south switch (for better visibility, maybe?)
  - Instead, manually jog to the center of the box
  - Instead, manually jog to one of the corners of the box
  - Additionally, do a second pass around the four switches to paranoically try to compensate for manual positioning error even more
  - Provide this functionality as a new CLI calibration command instead of changing the existing behavior of `p`
  - etc.

SyntaxColoring, 2020-02-18:
Adding `papi v2` tag because I don't think this is specific to APIv1.

sfoster1, 2020-04-27:
This will be fixed in the robot calibration overhaul.

closed 2020-04-27 by sfoster1