Opentrons/opentrons #3986: After a factory reset, and before the first successful GUI pipette calibration, pipettes are way too low

SyntaxColoring, 2019-08-30:
# Overview

Without having already established good pipette calibrations through the pipette calibration GUI, pipettes will be positioned too low. They will be 

This occurs after CLI calibration on fresh SD cards. It also occurs after doing a tip length factory reset through the GUI.

We've seen this issue across multiple OT-2s.

# Reproducing

1. Feed an OT-2 a fresh SD card. We've seen this on a few different system versions, but the  VERSION.json corresponding to the sdcard.img I used is: `{"buildroot_version": "v1.0.3-1-gc4bf505b1", "buildroot_sha": "c4bf505b1562e9d9a7a3544a19cd16151a3b23f6", "buildroot_branch": "opentrons-develop", "buildroot_buildid": "opentrons-buildroot:61c73d8b-97f6-4bc9-a3b6-6ef4b482b286", "build_type": "dev", "opentrons_api_version": "3.11.3", "opentrons_api_sha": "b88c73ba9a79758cf130d2c24d8700aa5c13bcc8", "opentrons_api_branch": "edge", "update_server_version": "3.11.3", "update_server_sha": "b88c73ba9a79758cf130d2c24d8700aa5c13bcc8", "update_server_branch": "edge"}`
2. Do CLI calibration. The tip probe stage (`p`) may fail initially, but after multiple `p`+`spacebar` iterations with manual "helping," it will stabilize and succeed. See also #3985.
3. Reboot and upload a protocol through the GUI. I used one with two P300s. When you do pipette calibration, the pipettes will be too low, despite having done a successful tip probe in CLI calibration.
4. After many iterations with manual intervention, pipette calibration will eventually stabilize and succeed.
5. Do a **Reset** > **Tip Length**. Observe you're now back at step 3.

# Reading Material

I've attached a .zip with a history of JSON files.

  1. `robot_settings_0_after_cli_calibration.json` and `deck_calibration_0_after_cli_calibration`.json are from immediately after step 2, above.
  2. `robot_settings_1_after_much_pipette_calibration_finagling.json` is immediately after step 4, above.
  3. `robot_settings_2_after_factory_reset.json` is immediately after step 5, above.

opatel0, 2019-09-19:
In some cases, this can result in the z-switch on the tip probe being triggered too early during tip length calibration, which results in the error message below
[image]

opatel0, 2019-09-19:
Ran into a strange scenario using a workaround to resolve this issue.

Right now, when users run into this issue, we manually change the tip length data in robot_settings.json, adding 10-15mm to provide a margin of safety so that the next time the robot does tip length calibration, it will be able to do so successfully and independently.

Today when I did this with a user, the robot was able to successfully complete tip length calibration after making the change to tip length, but ran into the issue again on a subsequent run with the same pipette and same tips. 

The first time we changed robot_settings.json, there was only one key/value pair for the p1000:

"p1000_single_v1.4": xx.xxxxx

(I don't remember the original value, apologies)

When the user ran into the issue again, when we opened robot_settings.json, there were two separate key/value pairs:
"p1000_single_v1": xx.xxxxx
"p1000_single_v1.4": xx.xxxxx

The v1 had a corrupt data point that was causing the issue. We manually changed the value for v1 and resolved the issue.

Hope this info helps!

PR #4119 (2019-09-27): fix(api): Fix miscellaneous bugs in deck CLI
## overview

closes #3986 and closes #4114. This PR fixes some gross movement issues that happened in deck calibration on the api V1 system.

## changelog

- Calculate mount offset based on expected points with the corrected tip length of a particular pipette model
- Add the updated Z value from gantry calibration into the robot position tracker
- Move to points 1/2/3 with Z mount offset subtracted out

## review requests

Do CLI at least 2x in a row. Check that your data gets overwritten properly and that your tip probe center data is ~72-75.
closed 2019-09-27 by Laura-Danielle