Opentrons/opentrons #4002: Jupyter Notebook is using a different directory to store/retrieve saved protocols on robots with 3.11+ server version

ethanfjones, 2019-09-04:
#3369 

 ## overview

Protocols saved to your robot's Jupyter Notebook pre 3.11 are not easily accessible after updating to 3.11+. 

## current behavior

When accessing your robot's Jupyter Notebook it looks for protocols saved in: 

 `/var/lib/jupyter/notebooks`

Before updating the robot to 3.11+, the robot's Jupyter Notebook looked for protocols in:

`/data/user_storage/opentrons_data/jupyter`

After updating to 3.11+ the protocols I saved to Jupyter Notebook are no longer easily accessible. 

## expected behavior

The protocols I had saved to my robot before updating to 3.11+ should accessible when I launch Jupyter Notebooks.

nmih, 2019-09-06:
Agreed, it would have been nice to have had this change documented somewhere. For now I have just had to ssh into the machine (which also changed and needed and ssh key https://support.opentrons.com/en/articles/3203681-enabling-ssh-access-to-your-ot2) and move the files manually through the terminal.

sfoster1, 2019-09-10:
@nmih It actually wasn't an intended change - it's a bug, and one we're fixing in 3.12.0.

Upon update to 3.12.0, Jupyter notebooks and other files in Jupyter's filesystem created prior to 3.11.0 will appear again. If you created a file in 3.11.x with the same name as a file created previously to 3.11.x (for instance, the `Default.ipynb` notebook), then the one created prior to 3.11.0 will appear with a numerical suffix.

closed 2019-09-10 by sfoster1

nmih, 2019-09-11:
> @nmih It actually wasn't an intended change - it's a bug, and one we're fixing in 3.12.0.
>
> Upon update to 3.12.0, Jupyter notebooks and other files in Jupyter's filesystem created prior to 3.11.0 [...]

Thanks for the information!