Opentrons/opentrons #4055: PD: Unable to upload Protocol Designer v3.0 protocols

meatyziti, 2019-09-13:
## overview
Unable to upload protocols exported from Protocol Designer v3.0+. Receive the following error in the Opentrons App upon upload: `JSONDecodeError:Expecting value line 1 column (Char 0)`.

## current behavior
When uploading a protocol exported from Protocol Designer v3.0+, the Opentrons App throws the following error: 

`JSONDecodeError:Expecting value line 1 column (Char 0)`

This occurs with multiple PD protocols created with v3.0+. User does not receive error when uploading protocols created with previous version of PD (v1.1).

## steps to reproduce
Robot server/Opentrons App version 3.10.3+
Upload protocol created with Protocol Designer v3.0+

## expected behavior
Successfully upload protocol created with Protocol Designer regardless of the version of PD.

btmorr, 2019-09-13:
Hi @meatyziti , thanks for filing a ticket. I haven't seen this before, and have uploaded protocols exported by PD. Is it possible to share a copy of the protocol that you're trying to upload? Also, what operating system are you using, and what browser were you using with PD?

I'll do my best to figure out why it's not working in this case. Thank you!

mcous, 2020-01-17:
Closing because unable to reproduce on PD 3.0.7 and App/API 3.15.2.

@meatyziti if this continues to happen to you, please reach out to our support team with the specific protocol and we can see what's going on

closed 2020-01-17 by mcous