Opentrons/opentrons #4069: Component Library selects should use generic select

b-cooper, 2019-09-17:
## overview

In the interest of UX unification, we'd like to move towards a place where very naive, flexible, reusable components live in CL and are subsequently instantiated in use-case specific components across the codebase. 

Select boxes provide a good first case for unifying our UX across apps. Right now we have two components that integrate react-select's expanded select box functionality, `PipetteSelect` and `SelectField`. They duplicate a lot of customization (mostly styling) across their two definitions.  We should create a third component that acts as a thin wrapper of react-select with only the shared styling customizations plugged in by default. This component should be as flexible as possible given the baseline similarities in appearance between instances of it (read; passthrough props into different levels of react-select). 

For the first pass, this new component should be used for both `SelectField` and `PipetteSelect`. The larger followup should see all dropdowns across the app use an instance of this generic component (probably through a use-case specific wrapping component similar to `SelectField` and `PipetteSelect`).

mcous, 2020-01-17:
Marking as `cpx` because we have upcoming work on the Wi-Fi select menu in the app which uses a bespoke (and incorrect) implementation of `react-select`

mcous, 2020-11-19:
Calling this one closed by the addition of the [`<Select>` component](https://s3-us-west-2.amazonaws.com/opentrons-components/edge/index.html#select) to the component library

closed 2020-11-19 by mcous