Opentrons/opentrons #4078: GUI for creation of Python protocols and custom hardware (Marlin) support

DerAndere1, 2019-09-18:
## overview

I am not a OT user, so this is low priority: A graphical protocol designer that outputs opentrons APIv2-compatible python protocols that can be edited manually in a second step. 

I am playing around with adding the above in rudimentary form to my toy project "GGCGen". The backend of GGCGen can translate to G-code (Marlin firmware compatible). This might help with teaching / promoting of opentrons API programming with Python: Students can run basic opentrons protocols on a 3D-printer with Marlin firmware which can be more entertaining than virtual simulation.

## current behavior
User has to choose between the great opentrons Python API or the intuitive Protocol Designer

## steps to reproduce
N.A.

## expected behavior
Graphical protocol designer that auto-generates python protocols, sends warnings for unreasonable input and allows editing of the pyhon code in a second step.

DerAndere1, 2019-09-18:
I put this to the archive for now as it is probably not so relevant for the Opentrons core project.

closed 2019-09-18 by DerAndere1

naikymen, 2022-12-16:
Hi! I just found this. The work you've done looks **awesome**.

A couple friends and I have been working on very similar stuff, and also posted in this repo a while ago: #11542

I was hoping there is a chance for us to collaborate. Please let me know if you want to have a chat :)

You can find me as `naikymen` at [the GOSH forum](https://forum.openhardware.science/u/naikymen/), [the PLR forum](https://forums.pylabrobot.org/u/naikymen/), and elsewhere.

Cheers!