Opentrons/opentrons #4122: Camera on OT2 not responding

prmunn, 2019-09-27:
## overview
Attempting to connect to camera on OT2 over wifi using Postman. 

## current behavior
Bug. 
Robot returns a message saying "picture not saved" and a status of "500 Internal Server Error"

## steps to reproduce
Open Opentrons app to find wireless IP. 192.168.0.232 in this case.
Open Postman and enter:  "192.168.0.232:31950/camera/picture"
Select "POST" and hit "SEND"

## expected behavior
Should return an image of the robot's deck.

mcous, 2019-09-27:
I... did not know we had this endpoint. Apparently it was added with https://github.com/Opentrons/opentrons/pull/1251. I'm not sure this is intended to be user-facing, but it's in code so we should make sure it works.

`take_picture` in `api/src/opentrons/server/endpoints/control.py` uses a relative path, which in turn likely relies on the cwd. Best guess: the filesytem changes in the buildroot switch exposed the brittleness of this handler and broke it.

Stuff to do:

- Fix the endpoint
- Ensure tests for `POST /camera/picture` are in the release QC tests

sfoster1, 2019-10-02:
We've identified the bug and it will be fixed in 3.13.0

PR #4163 (2019-10-02): fix(api): Put camera images in tempdirs
And not in the cwd of the server.

Closes #4122

Testing: Put this on a robot and get a picture with `curl -X POST $(robot ip):31950/camera/picture > image.jpg; open image.jpg`
commit 1411da83 (2019-10-02): fix(api): Put camera images in tempdirs (#4163)

And not in the cwd of the server.

Closes #4122
closed 2019-10-02 by Laura-Danielle