Opentrons/opentrons #4164: unexpected behaviour of pipette.delay()

pietrosormanni, 2019-10-02:
## overview

1- unexpected behaviour of pipette.delay(seconds=X) - it does delay the run but not by the amount of X seconds specified in input (by less).

2- Also trying to print with robot.comment() the time since begins of execution prints the simulated time. This is probably expected but quite annoying for kinetic time-sensitive applications where it is important to know exactly at what time the reactants have been mixed.

## current behavior

1- pipette.delay(seconds=X) systematically triggers a delay of less than X seconds. Keeping a stopwatch and checking the robot moving by eye, it looks like the X seconds start when the pipette picks up the tip (or around that time) as opposed to when (after transferring and dropping the tip) the pipette.delay() function is called. In principle this may not be a problem but the documentation should clearly state what events triggers the pipette.delay(seconds=X) to start; it is clearly not when the function is called. In my example I call the function 3 times, is this a cumulative time since the start of the protocol?
I have the pdf API downloaded on Sep 27, 2019 and there is very little detail around the delay function, which is however essential for kinetic application (e.g. automating chemical reactions).
2- At the end of the script below, the call to robot.comment('Done, total time taken %d seconds' % (seconds_taken_so_far)) actually prints a few milliseconds of time, which is I guess the simulated time, but the actual robot runs takes about 4 minutes. It would be really good to have the ability to extract the actual time from the log.

## steps to reproduce

#Run the following simple script to reproduce the issue - keep a stopwatch to check the delay:

from opentrons import labware, instruments,robot
import time

tiprack300 = [labware.load('opentrons-tiprack-300ul', slot='11')]  

p300 = instruments.P300_Single(mount='left', tip_racks= tiprack300)

reaction_plate=labware.load('96-flat', slot='8')

delay_minutes=[1,1,2]

robot.comment('starting')
start=None
for j,delay in enumerate(delay_minutes) :
    step_start=time.time()
    if start is None : start=time.time()
    p300.pick_up_tip()
    p300.transfer(50,reaction_plate.well('A1'),reaction_plate.well('H12'),new_tip='never')
    p300.return_tip()
    step_seconds_taken=time.time()-step_start
    seconds_taken_so_far=time.time()-start
    robot.comment('carried out step at index %d - seconds_taken_so_far= %g' % (j,seconds_taken_so_far))
    wait_for_more_seconds=delay*60 - step_seconds_taken
    robot.comment('WAITING for %d seconds corresponding to %g minutes' % (round(wait_for_more_seconds),wait_for_more_seconds/60.)) 
    p300.delay( seconds = wait_for_more_seconds )
    robot.comment('') # new line

robot.comment('Done, total time taken %d seconds' % (seconds_taken_so_far))
robot.home()

#SOFTWARE VERSION I HAVE:
#Updated to latest version as of 2 Oct 2019. 
#Software Version: 3.12.0
#using a Mac App to connect to robot

## expected behavior

  when p300.delay(seconds=60) is called it should wait for exactly 60 seconds. Or the documentation of the API should clearly state from when the timer starts. (is it from last call of pick_up_tip()? Is it a cumulative time from the beginning of the run disregarding any other operation carried out by the robot?)

Overall it would be also nice to have the API state how to print the actual time in the log and not the simulated time. The command at the end robot.comment('Done, total time taken %d seconds' % (seconds_taken_so_far)) prints a few milliseconds of time but the protocol takes about 4 minutes to run.

rgatkinson, 2019-11-06:
FWIW, some amount of investigation leads me to suspect the following archeological analysis as the root cause of the issue (which I also observe):

* Pipette.delay is the following:
```
        self.robot.pause()
        _sleep(seconds)
        self.robot.resume()
```

* ``robot.pause()`` calls ``pause()`` on its driver (SmoothieDriver_3_0_0), which just clears a flag:
```
    def pause(self):
        if not self.simulating:
            self.run_flag.clear()
```

* This flag is checked by the driver at the _beginning_ of new commands (``move()``, ``home()``) to determine whether to proceed. However, if an asynchronous command (which they all seem to be) is _currently_ in-progress when ``pause()`` is called, ``pause()`` can easily return before that command completes execution. Thus, some of the ``_sleep()`` overlaps with the finishing up of the command, leading to the observed behavior.

rgatkinson, 2019-11-06:
Also: I observe that the Smoothie driver has a ``delay()`` method, which sends a 'dwell' GCODE; I wonder why this is unused and ``sleep()`` used in its stead? It would seem to give exactly the timing control that is requested here.

rgatkinson, 2019-11-12:
FWIW: using the driver ``delay()`` directly appears in my experience to address the issue, though I have not yet made a rigorous timing comparison along the lines of the one that @pietrosormanni carried out.

```
def dwell(pip, seconds=0, minutes=0):
    # like delay() but synchronous with the back-end

    msg = 'Dwelling for {0}m {1}s'.format(minutes, seconds)

    minutes += int(seconds / 60)
    seconds = seconds % 60
    seconds += float(minutes * 60)

    # the do_publish() calls make the log look nicer, but otherwise can be omitted
    opentrons.commands.do_publish(pip.robot.broker, opentrons.commands.comment, f=dwell, when='before', res=None, meta=None, msg=msg)
    pip.robot.pause()
    pip.robot._driver.delay(seconds)
    pip.robot.resume()
    opentrons.commands.do_publish(pip.robot.broker, opentrons.commands.comment, f=dwell, when='after', res=None, meta=None, msg=msg)

```

sfoster1, 2019-12-10:
Sorry for the time it took to get back to you, @pietrosormanni and @rgatkinson ! I took a look at this using the example protocol on robot software 3.15.1. I timed it with a stopwatch, and I also inserted some logging calls into the protocol (this is a sneaky thing you can do with python logging, if you're willing to look at the robot logs. At the top of your protocol, in v1 or v2, add
```
import logging
log = logging.getLogger('opentrons.protocols')
```
and then later, you can use `log.info("my message")` and it will show up in the robot API logs. This is the [python standard logging package](https://docs.python.org/3/library/logging.html).

the full modified protocol is attached at the bottom of this post.)

My stop watch and the logged timing agreed pretty accurately, modulo the speed of my thumb:

Delay 1:
before the delay, logged: `Dec 10 22:16:20 restless-breeze opentrons-api[30627]: waiting for 38.656060218811035 from delay=1m taken=21.343939781188965`
after the delay: `Dec 10 22:16:59 restless-breeze opentrons-api[30627]: done waiting` 
The log timestamps (sorry for low res) show ~38.6 seconds,as did my hand timing.

Delay 2:
before the delay, logged: `Dec 10 22:17:18 restless-breeze opentrons-api[30627]: waiting for 40.977076053619385 from delay=1m taken=19.022923946380615`
after the delay: `Dec 10 22:17:59 restless-breeze opentrons-api[30627]: done waiting`
Timestamps and clock agreed to ~41s

Delay 3:
before the delay, logged: `Dec 10 22:18:18 restless-breeze opentrons-api[30627]: waiting for 101.04638934135437 from delay=2m taken=18.95361065864563`
after the delay: `Dec 10 22:19:59 restless-breeze opentrons-api[30627]: done waiting`
Logs and clock agreed at 1:41

That all looks pretty correct to me. The problem is, unless you know what the robot's doing exactly it can be hard to decide when an action is done; in particular, the run log updates (which entry is blue) aren't necessarily lining up perfectly - there's a delay as the message gets through the websocket. The logs may be the best method to handle this.

Systematically, the delays should be accurate because all they are is a `time.sleep()` call.

To address your other concerns, we definitely do want to make the run logs better. The way they work now is that they're scraped during the simulation - where we don't actually do that `time.sleep`, so as not to delay simulation. We capture logs then so we can present that forward looking view of the execution. We need to make them update based on runtime information, which would also make dynamic protocols act a lot better. Not sure when we'll be getting to that - we have a large refactor planned for this and other improvements around protocol information in the Opentrons App.

Hopefully the logging information can help you make sure the timing is accurate! I'm sorry I can't help further. I'll keep this ticket open for a while just in case.

Full modified protocol:

```python
from opentrons import labware, instruments,robot
import time
import logging
log = logging.getLogger('opentrons.protocol')

tiprack300 = [labware.load('opentrons-tiprack-300ul', slot='11')]

p300 = instruments.P300_Multi(mount='right', tip_racks= tiprack300)

reaction_plate=labware.load('96-flat', slot='8')

delay_minutes=[1,1,2]

robot.comment('starting')
start=None
for j,delay in enumerate(delay_minutes) :
    step_start=time.time()
    if start is None : start=time.time()
    p300.pick_up_tip()
    p300.transfer(50,reaction_plate.well('A1'),reaction_plate.well('H12'),new_tip='never')
    p300.return_tip()
    step_seconds_taken=time.time()-step_start
    seconds_taken_so_far=time.time()-start
    robot.comment('carried out step at index %d - seconds_taken_so_far= %g' % (j,seconds_taken_so_far))
    wait_for_more_seconds=delay*60 - step_seconds_taken
    log.info(f'waiting for {wait_for_more_seconds} from delay={delay}m taken={step_seconds_taken}')
    robot.comment('WAITING for %d seconds corresponding to %g minutes' % (round(wait_for_more_seconds),wait_for_more_seconds/60.))
    then = time.time()
    p300.delay( seconds = wait_for_more_seconds )
    now = time.time()
    log.info(f'done waiting after {now-then}')
    robot.comment('') # new line

robot.comment('Done, total time taken %d seconds' % (seconds_taken_so_far))
robot.home()
```

rgatkinson, 2019-12-10:
Thanks for the thoughts; I'll have a look a bit deeper when I get a minute.

But why not just use the 'dwell' gcode? It's simple and clean. That would, with certainty, make sure that the delay was one experienced by the executing protocol, and we wouldn't have to reason and think about whether the asynchronies in the upper layers get cancelled out so, in the end, using sleep() would have the same effect?

sfoster1, 2019-12-11:
That's true - chalk it up to gcode as a second language :). I don't think it would avoid uncertainties, though, because we don't really use command queuing on the gcode level - just about every G0 is followed up with M400. So you'd still have to account for system latencies.

pietrosormanni, 2019-12-11:
Thank you both very much for the helpful suggestions. 

I guess the key thing in Seth's code is to know exactly what operation triggers the start of the delay timer (e.g. f'waiting for {wait_for_more_seconds} from delay={delay}m taken={step_seconds_taken}' we calculate step_seconds_taken from the beginning of the loop, but in a more complicated robot operations, what would target the real beginning of this timer?). Is it the pick_up_tip()?

Thank you again,
Pietro

rgatkinson, 2019-12-11:
"I don't think it would avoid uncertainties...just about every..."

That's the point really, I guess: if we use the dwell gcode, we don't even have to ask this question, so we won't overlook something (which it seems we are doing now, though we haven't found the precise edge case yet). I've been using dwell myself for the last month: it definitely works, reliably, and I can't see any downside. Can we make the switch?

Dwell also gives a precise answer to @pietrosormanni's question. To wit: if a protocol w/o dwell or delay takes t0 seconds, then adding a "dwell t seconds" will cause it to exactly take the desired t0 + t seconds, as enforced by the smoothie firmware, since the firmware simply sits and waits when it serially enounters the dwell (irrespective of what's happening in python or anywhere else). And the smoothie is _all about_ precise timing and movement, right?

I really think we should be using the dwell gcode rather than trying to simultate it in python (which is what we're doing now).

sfoster1, 2019-12-11:
@rgatkinson  When the system executes a protocol, it calls python's `exec` on your protocol file (if it is v1) or calls the `run` function you defined in your protocol - all python. As your protocol executes, it calls all those python functions. Some of them don't involve the smoothie - like loading labware. The ones that _do_ involve the smoothie get translated into gcode and sent to the smoothie, and then wait for the command to entirely complete before returning. So you have an execution for `move` and then `delay` that looks like this:

```
protocol                                python internals            serial link      smoothie
instr.move_to(wellplate['A1'])
                 | -------------------> move to (x=10, y=2, z=5)
                                                             |--> G0X10Y2Z200.0 --> enqueues move 
                                                             |       ok <---------- move enqueued, executing
                                                             |      M400 ---------> await move complete
                                                             | <---- ok <---------- move complete                                                                       
                                         returns <-----------|
returns------------------------------------|
delay(seconds=60)
                 | -------------------> time.sleep(60)
                 <--------------------- returns
```

Because we don't queue, the only difference between this code and code that delays using `dwell` is you'd add that right side of the diagram to the `delay` call as well.

So it's less that we're simulating the delay in python and more that given the system as a whole, with both the raspberry pi and the smoothie, and the smoothie integrated into the python stack synchronously, we might as well do the delay on the python side.

@pietrosormanni The specific answer is, the delay timer starts counting as soon as python calls `sleep`, which is a couple function calls down the stack from calling `delay`. The run log may update to indicate that the delay is currently ongoing some couple hundred milliseconds later. *The delay begins as soon as the previous line of the protocol completes and the interpreter get to the delay()` call*.

The question of when the delay happens relative to other actions kind of depends on the other action. For instance, if there's a delay right after a `drop_tip()`, there will be about 8 seconds between when the tip falls off the pipette and the `delay` begins - that's because in those 8 seconds the plunger is homing after possibly skipping when dropping the tip, which is considered part of the drop tip process and so done in the `drop_tip` call, but isn't externally visible.

rgatkinson, 2019-12-11:
I understand the intended flow.

You seem to have entirely missed my point. I storngly disagree what we 'might as well delay on the python side'. That's _intrisically_ more fragile.

mcous, 2019-12-18:
@pietrosormanni several of us have tried your protocol on a few robots and found that `pipette.delay` (`ProtocolContext::delay` in Protocol APIv2) delayed for exactly as many seconds as were specified. Accordingly, I'm closing this issue. I instrumented your protocol with some simple `prints` and ran it via `opentrons_execute` on the robot directly. I've attached the print log below showing the times (and I confirmed them by hand with a timer). As @sfoster1 said, to answer your question in the original ticket: **the delay starts when the Python executor hits the `pipette.delay` line**.

**We definitely agree that the run log not reflecting reality is painful and confusing**, and I appreciate you bringing it up. We've been stuck with an exceptionally poor system + data model for linking simulated commands to actually run commands that we're actively unwinding so we can implement a run experience that actually works. #2047 is the best ticket we have right now for tracking that work, and your ticket here is officially part of the record for why we need to fix it.  

@rgatkinson relying on Smoothie for "wait an amount of time" does not align with our architectural goals for the software for the reasons Seth listed above. The Smoothie board is not the OT-2, it is a single peripheral used by the OT-2 to accomplish pipette motor movements. The Python server is the business logic of the robot, and any logic that can be consolidated to the Python server will be. In my experience, functionality that depends on one system is more predictable and stable than functionality that depends two systems.

Logs of delay testing:

```
Using a deprecated constructor for p300_single_v2.0
 starting
 Picking up tip from well A1 in "11"
alarm/error: command=M907 A0.8 B0.05 C0.05 X0.3 Y0.3 Z0.1 G4P0.005 G0A201, resp=ALARM: Hard limit +A
 Transferring 50.0 from well A1 in "8" to well H12 in "8"
	 Aspirating 50.0 uL from well A1 in "8" at 1 speed
	 Dispensing 50.0 uL into well H12 in "8" at 1 speed
 Returning tip
v1 location: (<Deck><Slot 11><Container opentrons-tiprack-300ul><Well A1>, (x=1.75, y=1.75, z=35.50))
	 Dropping tip into well A1 in "11"
wait_for_more_seconds: 32.40715146064758, time: 1576695742.5318258
 Delaying for 0 minutes and 32.40715146064758 seconds
	 Pausing robot operation
	 Resuming robot operation
delay done: time: 1576695774.9727697
delay done
 Picking up tip from well B1 in "11"
 Transferring 50.0 from well A1 in "8" to well H12 in "8"
	 Aspirating 50.0 uL from well A1 in "8" at 1 speed
	 Dispensing 50.0 uL into well H12 in "8" at 1 speed
 Returning tip
v1 location: (<Deck><Slot 11><Container opentrons-tiprack-300ul><Well B1>, (x=1.75, y=1.75, z=35.50))
	 Dropping tip into well B1 in "11"
wait_for_more_seconds: 36.492067098617554, time: 1576695798.4809506
 Delaying for 0 minutes and 36.492067098617554 seconds
	 Pausing robot operation
	 Resuming robot operation
delay done: time: 1576695835.0239847
delay done
 Picking up tip from well C1 in "11"
 Transferring 50.0 from well A1 in "8" to well H12 in "8"
	 Aspirating 50.0 uL from well A1 in "8" at 1 speed
	 Dispensing 50.0 uL into well H12 in "8" at 1 speed
 Returning tip
v1 location: (<Deck><Slot 11><Container opentrons-tiprack-300ul><Well C1>, (x=1.75, y=1.75, z=35.50))
	 Dropping tip into well C1 in "11"
wait_for_more_seconds: 96.554368019104, time: 1576695858.4698734
 Delaying for 0 minutes and 96.554368019104 seconds
	 Pausing robot operation
	 Resuming robot operation
delay done: time: 1576695955.0660894
delay done
 Done, total time taken 143 seconds
```

closed 2019-12-18 by mcous

rgatkinson, 2019-12-18:
@mcous : Thanks for the thoughtful explanation.