Opentrons/opentrons #4192: Is there a way to simulate OT1 protocols?

joshuak94, 2019-10-09:
If not, I guess this would be a feature request. Maybe some sort of flag in `opentrons_simulate` that lets it switch between OT1 and OT2? Currently, I get warnings and errors about deprecated parameters (`axis` in `Pipette`), but as far as I can tell, OT1 uses `axis` and OT2 uses `mount`.

mcous, 2019-10-11:
Hi @joshuak94, thanks for the question and request! We don't have engineering resources to devote to adding OT-1 support to the current version of the `opentrons_simulate` script at this time.

However, the OT-1 runs on the 2.x line of the `opentrons` pip package, and that version of the software exposed a `robot.commands` method to return a list of all commands executed. The following works on my machine to get a protocol simulation:

1. Create and activate a Python virtual environment of Python 3.5
2. `pip install opentrons==2.5.2`
3. `python ./my-protocol.py`

In your protocol:

1. Ensure you do not call `robot.connect()`
2. Add a `print` statement that prints `robot.commands()`

For example:

```py
from opentrons import robot
from opentrons import containers, instruments

tiprack = containers.load(
    'tiprack-200ul',  # container type
    'A1',             # slot
    'tiprack'         # user-defined name
)

plate = containers.load('96-flat', 'B1', 'plate')

p200 = instruments.Pipette(
    axis="b",
    max_volume=200
)

p200.pick_up_tip(tiprack[0])

for i in range(95):
    p200.aspirate(100, plate[i])
    p200.dispense(plate[i + 1])

p200.return_tip()

if robot.is_simulating():
    print('\n'.join(robot.commands()))
```

Does this work for your needs?

mcous, 2019-11-25:
Closing due to lack of response and because I _think_ my response is sufficient for the needs of the OP. Please let me know if that's not the case!

closed 2019-11-25 by mcous

JoseTomasTocino, 2020-03-30:
Quick question @mcous. I'm right in the same spot as the OP, I've been tasked to make some protocols for an OT-1 that I don't even have yet, so I have to simulate them somehow. I've tried your example (running `opentrons==2.5.2`) and it works and prints every step. My question is, is that what the current `opentrons_simulate` application does? Is that the extent of the simulation I can get? I guess it is, I mean there's not much more to it than showing what steps it would execute, but maybe I'm missing something.