Opentrons/opentrons #4200: Labware with too many rows runs out of alphabet

SyntaxColoring, 2019-10-09:
This is only a practical issue if labware can ever span multiple slots.

If labware has a lot of rows – more rows than can physically fit in one slot – the row labels will overflow the alphabet. They're just incremented naively by ASCII value.

https://github.com/Opentrons/opentrons/blob/a46f921ee5972730f5022dd78e78ddb511df64f6/api/src/opentrons/legacy_api/containers/__init__.py#L116

https://github.com/Opentrons/opentrons/blob/f8603a6c412fff1e3c51dcb731a34326c7b538dd/shared-data/js/helpers/index.js#L78

Bonus fun: if you try to do anything in the row labeled "{", something in the stack breaks because it tries to look for a matching "}".  🙃

IanLondon, 2019-10-11:
This is an issue if we decide to support 1536 well plates (do fit in 1 slot). They have 32 rows, first A-Z, then the remaining 6 rows are conventionally named AA, AB, ..., AF.

I don't know if there's a use case for non-alphabetical rows, but it's possible to double up the letters like 1536 plates do. We don't yet support that feature across the stack.

SyntaxColoring, 2021-06-03:
We're deprioritizing this because we think that, because of physical accuracy/precision limitations (especially with curved pipette tips), no one will try using labware with this many rows.

If this assumption turns out to be wrong, we're happy to revisit this.

closed 2022-09-30 by bndo