Opentrons/opentrons #4202: Protocol won't upload if robot and app versions are out of sync

ahiuchingau, 2019-10-10:
## overview
This is an issue we began to see in 3.12.0. If the app is updated to 3.13 and the robot is still on 3.12, the protocol cannot be uploaded correctly and the app shows the following message:

[image]

## expected behavior
There should be certain limitations if the app and the robot are out of sync, but should not affect user from uploading and running protocols on the robot.

SyntaxColoring, 2019-11-15:
Reopening this because it's been reported again on App v3.14.0 and server v3.10.1.

IanLondon, 2019-11-15:
https://github.com/Opentrons/opentrons/pull/4205/files#diff-4126c2cc2b8a3ba5e347a5700ed5db05R162 this might have to also handle a case where `error.name` is 'TypeError'???

SyntaxColoring, 2019-11-18:
Reported also on App v3.13.2 and server v.3.10.1.

mcous, 2019-11-25:
Upon further investigation, #4203 did not actually fix the issue in production. This was the result of the JS minifier used in production (but not dev) re-writing the name of our RPC client's internal `RemoteError` to a mangled shortened name

https://github.com/Opentrons/opentrons/blob/fb5d160e9e33b55e529cf5d51f6cd643710e50dd/app/src/rpc/remote-error.js#L1-L10

Fix incoming

PR #4203 (2019-10-10): fix(app): fix broken back-compat for pre-3.13.x robots
## overview

Closes #4202 

In #4099, I added a new argument to `Session.create` to support bundle execution, but missed the case where the App version was ahead of the robot server. This makes App backwards-compatible when calling `Session.create` via RPC.

## changelog

## review requests

Push v3.12.0 on to a robot and check:
- [ ] Robot no longer raises error described in #4202 
- [ ] Protocols with errors should show the error as expected (and not swallow it)

- [ ] .py, .json, and bundle execution should still work the same when conn [... cut]
commit 9243a8db (2019-10-10): fix(app): fix broken back-compat for pre-3.13.x robots (#4203)

closes #4202
PR #4205 (2019-10-11): Release 3.13.2 (hotfix)
## overview

Bumps to 3.13.2, including the hotfix which closes #4202 

## DO NOT SQUASH MERGE

Once approved, do non-squash merge to `master` from command line
PR #4509 (2019-11-25): fix(app): Call correct create method depending on protocol
## overview

This PR refactors how the RPC API client calls `session_manager.create` in order to:

- Support using custom labware in Python protocols (see #4490)
- Fix #4202 for real this time

## changelog

- Call the correct `session_manager.create...` method depending on the protocol and what methods the remote `session_manager` has available
    - If ZIP, call `session_manager.create_from_bundle` if available, else `create(..., true)`
    - If Python, call `create_with_extra_labware` or `create` depending on if the user has custom  [... cut]
closed 2019-11-25 by mcous