Opentrons/opentrons #4219: App allows user to

SyntaxColoring, 2019-10-14:
## overview

It's easy to accidentally contaminate a pipette by aspirating without first picking up a tip. When this happens, we need to ship customers replacement pipettes.

## current behavior

Protocols that include atomic liquid handling commands, like `aspirate()`, `mix()`, etc. are permitted to run even if they forget to `pick_up_tip()`. The pipette nozzle will move to the well and happily aspirate liquid into itself, contaminating it.

Warning messages are printed during local simulation with `opentrons_simulate`. However, no such messages appear before running a protocol through the app. So, at least some part of the software knows that the protocol is problematic, but that knowledge isn't making its way to the user.

## expected behavior

The user should at least be warned before running such a protocol.

* In the run log, warning messages could be printed as they are with `opentrons_simulate`.
* A warning dialog could confirm the user's intentions before allowing the protocol to run.
* etc.

SyntaxColoring, 2019-10-14:
See also #3535 and #3536.

mcous, 2019-10-15:
Since the app has very little to do with protocol execution (except for uploading the contents of the file and issuing "start", "pause", and "cancel" controls), and based on our real life conversation, I've tweaked the name and categorization of this issue to reflect that these actions should be blocked at the robot side

btmorr, 2019-12-03:
Fix merged to 3.15.0 release branch. Will be merged back into `edge` after release

commit cfb4d794 (2019-12-03): fix(api): prevent liquid handling without a tip

Fixes #4219
commit 6a4a15b7 (2019-12-03): fix(api): prevent liquid handling without a tip

Fixes #4219
PR #4528 (2019-12-03): fix(api): prevent liquid handling without a tip
## overview

Prevent liquid handling actions (aspirate/dispense) if the pipette does not have a tip. Fixes #4219 

## review requests

Example protocol (the basic protocol from the v2 docs) with the pick_up_tip call commented out:

```
from opentrons import protocol_api

# metadata
metadata = {
    'protocolName': 'My Protocol',
    'author': 'Name <email@address.com>',
    'description': 'Simple protocol to get started using OT2',
    'apiLevel': '2.0'
}

# protocol run function. the part after the colon lets your editor know
# where to look for [... cut]
commit e1724ab3 (2019-12-03): fix(api): prevent liquid handling without a tip (#4528)

* fix(api): prevent liquid handling without a tip. Fixes #4219
closed 2019-12-03 by btmorr