Opentrons/opentrons #4221: RFC: shared-data: define shape of module definitions to account for multiple versions of modules

sfoster1, 2019-10-15:
## overview

We are developing a second generation of the Temperature Module and Magnetic Module, with changes in geometry to avoid interference with labware placed in adjacent slots. 

These modules have the same functionality as their gen1 equivalents, but different identifiers and different geometry.

To provide backwards compatibility and ease of use in the protocol API, similar rules need to be enforced about specifying them by name: a protocol that specifies "magnetic module" should still function when a gen2 magnetic module is attached, but a protocol that specifies "temperature module gen2" should not work when a temperature module gen1 is attached (since the user presumably has a reason for specifying the gen2). 

This requires changes to the shape of our module definitions to allow linking the two generations of modules.

Please comment on this ticket with questions, comments, or suggestions for the new shape of the modules definition.

## proposed shape

- A new optional key in the per-module object called `backCompatName` with a mandatory string value that MUST be a `loadName` of another item in the top level object
- when `ProtocolContext.load_module` or `modules.load` is called, if the specified name matches a `backCompatName` of an attached module it is accepted

btmorr, 2019-10-15:
For reference, here are the current definitions: https://github.com/Opentrons/opentrons/blob/edge/shared-data/module/definitions/1.json

sfoster1, 2020-01-16:
In addition to the proposed shape above, i'd suggest:

validSlots: object with keys of slot names. value is an object with at least the key "transform", whose value is a 2d affine transform that should be applied to the labwareOffset when the module is loaded in that slot.

We currently only restrict module loads for the thermocycler, and only in code; so we can also have the implicit definition that if a module is loaded into a slot without a transform, the transform is identity and it's allowed.

closed 2020-01-16 by sfoster1